ros/rosdistro #13378: Installation Kinetic on MacOS Sierra failed while building 'opencv3'

aaramirezd, 2016-12-09:
MacBook-Pro-de-Andres-2:install andresramirez$ make -j4 -l4
[  3%] Built target zlib
[ 12%] Built target libjpeg
[ 12%] Built target libjasper
[ 17%] Built target libwebp
[ 17%] Automatic moc for target opencv_highgui
[ 17%] Automatic moc for target opencv_cvv
[ 18%] Built target libpng
[ 24%] Built target libtiff
[ 24%] Built target opencv_highgui_automoc
[ 24%] Built target opencv_cvv_automoc
[ 32%] Built target IlmImf
Scanning dependencies of target opencv_core
[ 32%] Building CXX object modules/core/CMakeFiles/opencv_core.dir/src/system.cpp.o
[ 34%] Linking CXX shared library ../../lib/libopencv_core3.dylib
[ 39%] Built target opencv_core
[ 39%] Linking CXX shared library ../../lib/libopencv_hdf3.dylib
[ 39%] Linking CXX shared library ../../lib/libopencv_flann3.dylib
[ 39%] Linking CXX shared library ../../lib/libopencv_ml3.dylib
[ 39%] Built target opencv_hdf
[ 39%] Built target opencv_flann
[ 39%] Linking CXX shared library ../../lib/libopencv_surface_matching3.dylib
[ 41%] Built target opencv_ml
[ 41%] Linking CXX shared library ../../lib/libopencv_viz3.dylib
[ 41%] Built target opencv_surface_matching
Undefined symbols for architecture x86_64:
  "vtkOBJReader::New()", referenced from:
      cv::viz::Mesh::load(cv::String const&, int) in types.cpp.o
      cv::viz::readCloud(cv::String const&, cv::_OutputArray const&, cv::_OutputArray const&) in vizcore.cpp.o
  "vtkSTLReader::New()", referenced from:
      cv::viz::readCloud(cv::String const&, cv::_OutputArray const&, cv::_OutputArray const&) in vizcore.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libopencv_viz3.3.1.0.dylib] Error 1
make[1]: *** [modules/viz/CMakeFiles/opencv_viz.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 41%] Linking CXX shared library ../../lib/libopencv_imgproc3.dylib
[ 46%] Built target opencv_imgproc
make: *** [all] Error 2

I did not find issues about it, 

Thank,
Andrés

vrabaud, 2016-12-10:
Which version did you install? I upgraded it recently to fix https://github.com/ros-gbp/opencv3-release/issues/6
Please open up a new issue in that repo if it still fails with  3.1.0-18.
Also, please look upstream if it has been fixed and in which patch. Thx !

vrabaud, 2016-12-11:
From what I read online, make sure that your brew software is up to date.

vrabaud, 2016-12-13:
@wjwwood , can you please try with the latest and close if it works for you? Thx.

wjwwood, 2016-12-22:
I commented on https://github.com/ros-gbp/opencv3-release/issues/7#issuecomment-268666400, let's close this one and handle it there.

closed 2016-12-22 by wjwwood