ros/rosdistro #13544: rosdep can't resolve lua5.2-dev on Xenial

dmiklic, 2017-01-09:
On Ubuntu Xenial with ROS Kinetic installed:

```
$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://github.com/ros/rosdistro/raw/master/rosdep/osx-homebrew.yaml
Hit https://github.com/amor-ros-pkg/rosdistro/raw/master/rosdep/base.yaml
Hit https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml
Hit https://github.com/ros/rosdistro/raw/master/rosdep/ruby.yaml
Hit https://github.com/ros/rosdistro/raw/master/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"
Add distro "hydro"
Add distro "indigo"
Add distro "jade"
Add distro "kinetic"
updated cache in /home/damjan/.ros/rosdep/sources.cache
$ rosdep resolve lua5.2-dev
ERROR: no rosdep rule for 'lua5.2-dev'
```

Although I can see the appropriate definition in [base.yaml](https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml):
```
lua5.2-dev:
  debian: [liblua5.2-dev]
  fedora: [lua]
ubuntu: [liblua5.2-dev]
```

dirk-thomas, 2017-01-09:
Works for me on Xenial without a problem:

```
$ rosdep resolve lua5.2-dev
#apt
liblua5.2-dev
```

tfoote, 2017-01-09:
It resolves for me just fine as well.  Can you show the output of `rosdep resolve lua5.2-dev -v` and `rosdep where-defined lua5.2-dev`? I see at least one non-standard rosdep source. Can you make sure that's not interfering?

dmiklic, 2017-03-09:
Sorry for not following up sooner. Here is the output.

```
$ rosdep where-defined lua5.2-dev
ERROR: cannot find definition(s) for [lua5.2-dev]
```
and
```
$ rosdep resolve lua5.2-dev
using matcher with tags [kinetic, ubuntu, xenial]
loading cached data source:
	https://github.com/ros/rosdistro/raw/master/rosdep/osx-homebrew.yaml
	/home/damjan/.ros/rosdep/sources.cache/5f44c73fb5a0e187f088839519d453ff8aa3699e.pickle
loading cached data source:
	https://github.com/amor-ros-pkg/rosdistro/raw/master/rosdep/base.yaml
	/home/damjan/.ros/rosdep/sources.cache/81ec6529b98b81dc74cac0568c94f6e7c0984228.pickle
loading cached data source:
	https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml
	/home/damjan/.ros/rosdep/sources.cache/f6f4ef95664e373cd4754501337fa217f5b55d91.pickle
loading cached data source:
	https://github.com/ros/rosdistro/raw/master/rosdep/ruby.yaml
	/home/damjan/.ros/rosdep/sources.cache/3db3f31778bf426ad9e41bef6144c78168b3c623.pickle
loading cached data source:
	https://github.com/ros/rosdistro/raw/master/releases/fuerte.yaml
	/home/damjan/.ros/rosdep/sources.cache/82cbc7008b5117bcc6ce794832659e4f7763d2db.pickle
loading cached data source:
	https://raw.githubusercontent.com/ros/rosdistro/master/groovy/distribution.yaml
	/home/damjan/.ros/rosdep/sources.cache/a75d3a0d86182afb2eb92f6c5473bf18db157d0e.pickle
loading cached data source:
	https://raw.githubusercontent.com/ros/rosdistro/master/hydro/distribution.yaml
	/home/damjan/.ros/rosdep/sources.cache/6a3d722dfe2d09a104a761d77901e00b2e7a540c.pickle
loading cached data source:
	https://raw.githubusercontent.com/ros/rosdistro/master/indigo/distribution.yaml
	/home/damjan/.ros/rosdep/sources.cache/f7622dadcf19914a1d2d3d8417ec62c3060c5d3e.pickle
loading cached data source:
	https://raw.githubusercontent.com/ros/rosdistro/master/jade/distribution.yaml
	/home/damjan/.ros/rosdep/sources.cache/0ce507007835947045d89dfa382199f6e83de08c.pickle
loading cached data source:
	https://raw.githubusercontent.com/ros/rosdistro/master/kinetic/distribution.yaml
	/home/damjan/.ros/rosdep/sources.cache/5d76dbb61bf5385acb00a1ffce292b2eb2e000a8.pickle
loaded 10 sources
4 sources match current tags
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] ignoring [python-nose] for os [fedora], already loaded
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] ignoring [python-nose] for os [gentoo], already loaded
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [macports] to [python-nose]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [opensuse] to [python-nose]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [slackware] to [python-nose]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [ubuntu] to [python-nose]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [arch] to [python-nose]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [debian] to [python-nose]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [osx] to [python-pydot]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] ignoring [python-pydot] for os [fedora], already loaded
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] ignoring [python-pydot] for os [gentoo], already loaded
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [macports] to [python-pydot]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [opensuse] to [python-pydot]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [slackware] to [python-pydot]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [ubuntu] to [python-pydot]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [arch] to [python-pydot]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [debian] to [python-pydot]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] ignoring [python-opencv] for os [ubuntu], already loaded
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [arch] to [python-opencv]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] ignoring [python-opencv] for os [fedora], already loaded
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [slackware] to [python-opencv]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [debian] to [python-opencv]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [ubuntu] to [python-vtk]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [arch] to [python-vtk]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] ignoring [python-vtk] for os [fedora], already loaded
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] adding rules for os [debian] to [python-vtk]
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] ignoring [python-sphinx] for os [fedora], already loaded
[https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml] ignoring [python-sphinx] for os [gentoo], already loaded
[... 83 lines cut ...]
set default installer for OS [opensuse]
[... cut ...]
view[*default*]: merging from cache of [https://github.com/amor-ros-pkg/rosdistro/raw/master/rosdep/base.yaml]
view[*default*]: merging from cache of [https://github.com/ros/rosdistro/raw/master/rosdep/python.yaml]
view[*default*]: merging from cache of [https://github.com/ros/rosdistro/raw/master/rosdep/ruby.yaml]
view[*default*]: merging from cache of [https://raw.githubusercontent.com/ros/rosdistro/master/kinetic/distribution.yaml]
view[*default*]: merging from cache of [sources.list]
view[*default*]: merging from cache of [<nodata>]
```

wjwwood, 2017-03-14:
Also "works" for me on OS X:

```
% rosdep resolve lua5.2-dev

ERROR: No definition of [lua5.2-dev] for OS [osx]

No definition of [lua5.2-dev] for OS [osx]
	rosdep key : lua5.2-dev
	OS name    : osx
	OS version : el capitan
	Data: debian:
- liblua5.2-dev
fedora:
- lua
ubuntu:
- liblua5.2-dev

% rosdep resolve --os=ubuntu:xenial lua5.2-dev
#apt
liblua5.2-dev
```

dirk-thomas, 2017-04-03:
I will close this for now since I assume the problem has been resolved in the meantime. If not please feel free to comment with more information and the ticket can be reopened.

closed 2017-04-03 by dirk-thomas