ros/rosdistro #13550: Lunar Kick-Off Checklist

mikaelarguedas, 2017-01-10:
This is a checklist of things to do for the Lunar kick-off (pre development and development):
- [x] Choose distribution name: Lunar
- [ ] commission a logo
- [ ] Add to REP3 with proposed platforms and minimum requirements 
- [ ] List on ROS wiki with expected release date (http://wiki.ros.org/Distributions)
- [ ] EOL distribution N-2 (Jade)
- [x] Add Ubuntu Z + Debian Stretch to bootstrap repository on repos.ros.org  https://github.com/ros-infrastructure/reprepro-updater/pull/35
- [ ] Add Ubuntu Z + Debian Stretch to buildfarm_deployment_config 
- [ ] Add Ubuntu Z + Debian Stretch to ros_buildfarm/common.py

---
- [ ] Import third party packages
  - [ ] `colladadom`

---
- [x] Add newly supported platforms to list of platforms for each Python package (in `stdeb.cfg`), in order of dependency (for ROS M, do this as soon as debs for new Ubuntu are available, in pre-alpha!):
  - [x] rospkg (uploaded, no need to rerelease): https://github.com/ros-infrastructure/rospkg/commit/60109b662b1cfd0e3fb8d59d1a72069cbbb631b9
  - [x] catkin_pkg (uploaded, no need to rerelease): https://github.com/ros-infrastructure/catkin_pkg/commit/0b418ae10260025016322e1f665c418bc726e241
  - [x] rosdistro (uploaded, no need to rerelease): https://github.com/ros-infrastructure/rosdistro/commit/1ba7c979d81a2d7358d608b15af233ffa4ec831f
  - [x] rosdep (uploaded, no need to rerelease):  https://github.com/ros-infrastructure/rosdep/commit/40067652cd0cde9d192e00788500d67124a30e27
  - [x] vcstools (uploaded, no need to rerelease): https://github.com/vcstools/vcstools/commit/a7b6e09e41145b5ade6251c9ae4562e8f0ae7935
  - [x] wstool (uploaded, no need to rerelease): https://github.com/vcstools/wstool/commit/84f0fe6497c88f7a95ab688aa2e39eb731d54e7b
  - [x] rosinstall (uploaded, no need to rerelease): https://github.com/vcstools/rosinstall/commit/5e12046c021b96d8e71f6c86ea427ec362fe6d1d
  - [x] bloom (uploaded, no need to rerelease):  https://github.com/ros-infrastructure/bloom/commit/9e35e0f53bad69e62f2e4df66d7830138d7ba5bb
  - [x] rosinstall_generator (uploaded, no need to rerelease):  https://github.com/ros-infrastructure/rosinstall_generator/commit/c89c9764e2fddd0d5f1f552f10282e6a9c43e591
  - [x] catkin_tools (uploaded, no need to rerelease): https://github.com/catkin/catkin_tools/commit/e5bb5d842dca57e489df5f06ec5fa71c3527324a
  - [x] vcstool: https://github.com/dirk-thomas/vcstool/commit/25a7c5461351906ea8de0522381b95321628da2d and https://github.com/dirk-thomas/vcstool/commit/953b294b4d379642168f1ba08a1f98b6322b8a2b
- [x] Use [`ros_release_python`](https://github.com/ros-infrastructure/ros_release_python) to make the currently released versions available in the new distros. Use the `--include` option

---
- [ ] create Docker images for the new platforms for all targets

---
- [ ] Create distribution files in ros/rosdistro
- [ ] add [ros_buildfarm_config](https://github.com/ros-infrastructure/ros_buildfarm_config) lunar entries
  - [ ] create new mailing list (https://groups.google.com/forum/#!forum/ros-buildfarm-kinetic) with exactly the same settings as the existing ones
- [ ] Disable jobs from build farm for EOL'ed distributions
- [ ] generate new management jobs (see https://github.com/ros-infrastructure/ros_buildfarm/blob/master/doc/deploy_configuration.rst)

---
- [ ] Remove old pre-release website link from pre-release website
- [ ] Add new column in [prerelease website](https://github.com/ros-infrastructure/prerelease_website) for the new distro with theme and placeholder image
  - [ ] Test prerelease process

---
- [ ] Add Lunar button to the [wiki](https://github.com/ros-infrastructure/roswiki)
- [ ] Add distribution to ROS browse tabs
- [ ] Edit wiki pages to contain proper content for the Kinetic tag.
- [ ] Add standard distribution wiki pages (planning, migration, etc...)
  - [ ] http://wiki.ros.org/lunar
  - [ ] http://wiki.ros.org/lunar/Planning
  - [ ] http://wiki.ros.org/lunar/Migration
  - [ ] http://wiki.ros.org/lunar/Planning/Maintenance
    -  Page made, update periodically with metapackage_status.py as packages get released
  - [ ] http://wiki.ros.org/lunar/Installation

---
*tag people once buildfarm is ready*
- [ ] Release `ros-base` packages into Lunar
  - [ ] catkin (dirk-thomas)
  - [ ] genmsg (dirk-thomas)
  - [ ] gencpp (dirk-thomas)
  - [ ] geneus (k-okada)
  - [ ] genlisp (dirk-thomas)
  - [ ] genpy (dirk-thomas)
  - [ ] gennodejs (chris-smith)
  - [ ] roscpp_core (dirk-thomas)
  - [ ] message_generation (dirk-thomas)
  - [ ] message_runtime (dirk-thomas)
  - [ ] rosbag_migration_rule (dirk-thomas)
  - [ ] std_msgs (dirk-thomas)
  - [ ] ros_comm_msgs (dirk-thomas)
  - [ ] cmake_modules (wjwwood)
  - [ ] rospack (dirk-thomas)
  - [ ] ros (dirk-thomas)
  - [ ] roslisp (dirk-thomas)
  - [ ] ros_comm (dirk-thomas)
  - [ ] rosconsole_bridge (dirk-thomas)
  - [ ] common_msgs (tfoote)
  - [ ] ros_tutorials (dirk-thomas)
  - [ ] actionlib (mikaelarguedas)
  - [ ] angles (tfoote)
  - [ ] bond_core (mikaelarguedas)
  - [ ] dynamic_reconfigure (mikaelarguedas)
  - [ ] class_loader (mikaelarguedas)
  - [ ] pluginlib (mikaelarguedas)
  - [ ] nodelet_core (mikaelarguedas)
  - [ ] common_tutorials (wjwwood)

---
- [ ] Announce availability of the Lunar build farm
- [ ] Update mirror documentation (see https://github.com/ros-infrastructure/mirror/commits/master for examples)
- [ ] Announce logo design and kick off t-shirt campaign

dirk-thomas, 2017-02-21:
@k-okada Can you please release `geneus` into Lunar.

@chris-smith Can you please release `gennodejs` into Lunar.

It can be just the latest Kinetic release so that we can continue releasing the ROS core packages quickly. Thanks!

chris-smith, 2017-02-21:
Done! ( through @IanTheEngineer )

dirk-thomas, 2017-02-21:
@chris-smith @IanTheEngineer Thank you guys for the super quick action :rocket:

dirk-thomas, 2017-02-21:
@wjwwood Can you please release `cmake_modules` into Lunar including changes specific to Lunar (e.g. ros/cmake_modules#42).

wjwwood, 2017-02-21:
Sure, here's the release pr: https://github.com/ros/rosdistro/pull/13957

dirk-thomas, 2017-02-21:
@wjwwood Thanks for the fast release.

k-okada, 2017-02-22:
`geneus` done -> https://github.com/ros/rosdistro/pull/13962

dirk-thomas, 2017-02-22:
@k-okada Thank you for the quick release :rocket:

mikaelarguedas, 2017-02-23:
@tfoote you should be able to release `common_msgs` now that `std_msgs` and below have been released

mikaelarguedas, 2017-02-23:
thanks @tfoote for the quick release!

mikaelarguedas, 2017-02-23:
@stonier  Can you please release `common_tutorials` into Lunar?

It can be just the latest Kinetic release. It's just to make sure that it actually builds properly.
Thanks!

dirk-thomas, 2017-02-23:
Actually it won't build atm because of some parent sourcedeb jobs being stuck. Therefore it's also not possible to test it using Debian packages of the dependencies.

stonier, 2017-02-23:
Aye, not going to build, but at least all the dependencies are there so it can be bloomed -> [ros/rosdistro#13999](https://github.com/ros/rosdistro/pull/13999).

When the deps are built, it should be right to go.

dirk-thomas, 2017-02-27:
I added a new bullet to the second block in the first comment: third-party packages need to be imported into the repo for all newly targeted platforms. One example I ran into which is currently missing: ros-infrastructure/ros_bootstrap_dependencies#5

Looking at http://repositories.ros.org/status_page/lunar_default.html other packages seem to be missing as well.

mikaelarguedas, 2017-03-06:
@dirk-thomas We decided to remove this bullet from this checklist and ask maintainers to request import of third party packages as they release. That allows us to flush out the packages that shouldn't be imported in the first place and to treat imports one at a time. So I'm going to remove this bullet and make it explicit on the release checklist ticket.

Thanks for pointing it out

mikaelarguedas, 2017-03-06:
@ros/ros_team I'm not sure what the purpose of the [Maintenance page](http://wiki.ros.org/kinetic/Planning/Maintenance) is, so could use some input on the following proposal.

I am considering redirecting the `Maintenance page` to the [status_page](http://repositories.ros.org/status_page/blocked_releases_lunar.html?q=id%3D%22metapackages%22&s=6&r=1) created for this purpose.

Pros:
- Up to date because generated periodically by the buildfarm
- Avoid having to generate manually the page every so often

Cons:
- It's not as eye friendly as the previous one and displays only the "Not released" packages

Alternative:
- generate a Maintenance page periodically during the beta cycle so that it stays roughly up to date
- leave it as is once all is green

Feedback welcome :)

dirk-thomas, 2017-03-06:
I think the new status page should replace the former wiki page.

tfoote, 2017-03-06:
+1 for replacing with the automated page. This wiki page was our manual approach to automating it. 

I'd suggest replacing the contents with a little bit of tutorials on how to use the status pages effectively. (aka a few sentances explaining the search queries and terms.)

wjwwood, 2017-03-06:
Yup, +1 for replacing it.

mikaelarguedas, 2017-03-06:
sounds good, thanks all for the quick feedback, I'll do that then.

mikaelarguedas, 2017-03-07:
@wjwwood can you confirm that "Add distribution to ROS browse tabs" was done in https://github.com/ros-infrastructure/rosbrowse/pull/4 ?

Should this be a single item on the [release checklist](https://github.com/ros/rosdistro/issues/14112) that consists in : 
 - adding the name of the next rosdistro (once ROS-M name has been decided)
 - updating the active one
and be completely removed from the Kickoff checklist?

wjwwood, 2017-03-08:
That's the right pr to update rosbrowse. I agree it should be part of the release and not the kick-off checklist because we usually announce the next name at the release of the current one and we shouldn't update lunar to be the default until it is released.

mikaelarguedas, 2017-03-08:
ok thanks for confirming, removing it from this checklist then

wjwwood, 2017-04-24:
🎉

closed 2017-04-24 by wjwwood