ros/rosdistro #13677: Lunar Check Wiki pages

mikaelarguedas, 2017-01-23:
This is copied from https://github.com/ros/rosdistro/issues/6878 and needs to be updated/completed before starting working on it.

This is a list of wiki pages that I think we should look at and update for Jade if needed:
- [ ] http://wiki.ros.org/actionlib_tutorials/Tutorials/RunningServerAndClientWithNodes at line 138
- [ ] http://wiki.ros.org/actionlib_tutorials/Tutorials/SimpleActionClient%28Threaded%29 at line 169
- [ ] http://wiki.ros.org/actionlib_tutorials/Tutorials/SimpleActionServer%28GoalCallbackMethod%29 at line 22
- [ ] http://wiki.ros.org/catkin at line 3
- [ ] http://wiki.ros.org/class_loader at line 4
- [ ] http://wiki.ros.org/compressed_depth_image_transport at line 29
- [ ] http://wiki.ros.org/compressed_image_transport at line 28
- [ ] http://wiki.ros.org/costmap_2d at line 53
- [ ] http://wiki.ros.org/cv_bridge/Tutorials/UsingCvBridgeToConvertBetweenROSImagesAndOpenCVImages at line 27
- [ ] http://wiki.ros.org/dynamic_reconfigure/Tutorials/SettingUpDynamicReconfigureForANode%28python%29 at line 28
- [ ] http://wiki.ros.org/gazebo at line 8
- [ ] http://wiki.ros.org/gazebo_ros_pkgs at line 21
- [ ] http://wiki.ros.org/image_proc/common at line 9
- [ ] http://wiki.ros.org/image_proc/cturtle at line 9
- [ ] http://wiki.ros.org/image_proc/diamondback at line 40
- [ ] http://wiki.ros.org/image_proc/electric at line 67
- [ ] http://wiki.ros.org/image_transport at line 112
- [ ] http://wiki.ros.org/image_transport_plugins at line 7
- [ ] http://wiki.ros.org/image_view/diamondback at line 182
- [ ] http://wiki.ros.org/interactive_marker_twist_server at line 10
- [ ] http://wiki.ros.org/mjpeg_server at line 9
- [ ] http://wiki.ros.org/navigation/Tutorials/RobotSetup/TF at line 65
- [ ] http://wiki.ros.org/opencv2 at line 12
- [ ] http://wiki.ros.org/os/OSX/Homebrew/troubleshooting at line 28
- [ ] http://wiki.ros.org/pcl/Tutorials at line 21
- [ ] http://wiki.ros.org/pluginlib/pluginlib_pregroovy at line 3
- [ ] http://wiki.ros.org/pluginlib/Tutorials/Writing%20and%20Using%20a%20Simple%20Plugin at line 37
- [ ] http://wiki.ros.org/pluginlib at line 5
- [ ] http://wiki.ros.org/ROS/Tutorials/rosdep at line 39
- [ ] http://wiki.ros.org/ROS/Tutorials/SetupEnvironment at line 9
- [ ] http://wiki.ros.org/rosbash at line 9
- [ ] http://wiki.ros.org/rosgraph_msgs at line 15
- [ ] http://wiki.ros.org/rviz/Tutorials/Librviz%3a%20Incorporating%20RViz%20into%20a%20Custom%20GUI at line 26
- [ ] http://wiki.ros.org/rviz/Tutorials/Markers%3a%20Basic%20Shapes at line 134
- [ ] http://wiki.ros.org/rviz/Tutorials/Plugins%3a%20New%20Display%20Type at line 26
- [ ] http://wiki.ros.org/rviz/Tutorials/Plugins%3a%20New%20Dockable%20Panel at line 26
- [ ] http://wiki.ros.org/rviz/Tutorials/Plugins%3a%20New%20Tool%20Type at line 26
- [ ] http://wiki.ros.org/simulator_gazebo/GazeboConfiguration at line 8
- [ ] http://wiki.ros.org/StackReleaseProcess at line 5
- [ ] http://wiki.ros.org/stage at line 7
- [ ] http://wiki.ros.org/stereo_image_proc/disparity_params at line 1
- [ ] http://wiki.ros.org/StyleGuide at line 439
- [ ] http://wiki.ros.org/teleop_twist_keyboard at line 11
- [ ] http://wiki.ros.org/test_macro at line 3
- [ ] http://wiki.ros.org/tf/Overview/Data%20Types at line 12
- [ ] http://wiki.ros.org/tf/Tutorials/Debugging%20tf%20problems at line 30
- [ ] http://wiki.ros.org/tf/Tutorials/Introduction%20to%20tf at line 157
- [ ] http://wiki.ros.org/tf/Tutorials/Using%20Stamped%20datatypes%20with%20tf%3a%3aMessageFilter at line 28
- [ ] http://wiki.ros.org/tf/Tutorials/Writing%20a%20tf%20broadcaster%20%28C%2b%2b%29 at line 68
- [ ] http://wiki.ros.org/tf/Tutorials/Writing%20a%20tf%20listener%20%28Python%29 at line 33
- [ ] http://wiki.ros.org/theora_image_transport at line 4
- [ ] http://wiki.ros.org/versionexample at line 8
- [ ] http://wiki.ros.org/visualization at line 3
- [ ] http://wiki.ros.org/WikiMacros at line 16

mikaelarguedas, 2017-05-23:
List of page using "rosversion kinetic"
- [ ] http://wiki.ros.org/rosjava/Tutorials
- [ ] http://wiki.ros.org/Installation/UbuntuARM/
- [ ] http://wiki.ros.org/kobuki/Tutorials/Installation/
- [ ] http://wiki.ros.org/kobuki/Tutorials/Installation/
- [ ] http://wiki.ros.org/kobuki/Tutorials/Installation/
- [ ] http://wiki.ros.org/kobuki/Tutorials/Installation/
- [ ] http://wiki.ros.org/android/Tutorials/
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/Xbot
- [ ] http://wiki.ros.org/Robots/MRP2
- [ ] http://wiki.ros.org/Robots/MRP2
- [ ] http://wiki.ros.org/Robots/MRP2
- [ ] http://wiki.ros.org/Installation/Ubuntu
- [ ] http://wiki.ros.org/Robots/qbmove
- [ ] http://wiki.ros.org/Robots/qbmove
- [ ] http://wiki.ros.org/Robots/qbmove
- [ ] http://wiki.ros.org/perception_oru
- [ ] http://wiki.ros.org/Robots/qbhand
- [ ] http://wiki.ros.org/Robots/qbhand
- [ ] http://wiki.ros.org/Robots/qbhand
- [ ] http://wiki.ros.org/Robots/qbhand
- [ ] http://wiki.ros.org/rviz/Tutorials/Librviz(3a20)Incorporating(20)RViz(20)into(20)a(20)Custom(20)GUI
- [ ] http://wiki.ros.org/ROSARIA/Tutorials/How(20)to(20)use(20)ROSARIA
- [ ] http://wiki.ros.org/ROSARIA/Tutorials/How(20)to(20)use(20)ROSARIA
- [ ] http://wiki.ros.org/ROSARIA/Tutorials/How(20)to(20)use(20)ROSARIA
- [ ] http://wiki.ros.org/ROSARIA/Tutorials/How(20)to(20)use(20)ROSARIA
- [ ] http://wiki.ros.org/ROSARIA/Tutorials/How(20)to(20)use(20)ROSARIA
- [ ] http://wiki.ros.org/ROSARIA/Tutorials/How(20)to(20)use(20)ROSARIA
- [ ] http://wiki.ros.org/ROSARIA/Tutorials/How(20)to(20)use(20)ROSARIA
- [ ] http://wiki.ros.org/ROSARIA/Tutorials/How(20)to(20)use(20)ROSARIA
- [ ] http://wiki.ros.org/ROSARIA/Tutorials/How(20)to(20)use(20)ROSARIA
- [ ] http://wiki.ros.org/ROSARIA/Tutorials/How(20)to(20)use(20)ROSARIA
- [ ] http://wiki.ros.org/ROSARIA/Tutorials/How(20)to(20)use(20)ROSARIA
- [ ] http://wiki.ros.org/motoman_experimental
- [ ] http://wiki.ros.org/hansbot/Tutorials
- [ ] http://wiki.ros.org/hansbot/Tutorials
- [ ] http://wiki.ros.org/hansbot/Tutorials
- [ ] http://wiki.ros.org/hansbot/Tutorials
- [ ] http://wiki.ros.org/hansbot/Tutorials
- [ ] http://wiki.ros.org/hansbot/Tutorials
- [ ] http://wiki.ros.org/hansbot/Tutorials
- [ ] http://wiki.ros.org/hansbot/Tutorials
- [ ] http://wiki.ros.org/hansbot/Tutorials
- [ ] http://wiki.ros.org/pioneer_bringup
- [ ] http://wiki.ros.org/pioneer_bringup
- [ ] http://wiki.ros.org/pioneer_bringup
- [ ] http://wiki.ros.org/pioneer_bringup
- [ ] http://wiki.ros.org/pioneer_bringup
- [ ] http://wiki.ros.org/pioneer_bringup
- [ ] http://wiki.ros.org/fanuc
- [ ] http://wiki.ros.org/visp
- [ ] http://wiki.ros.org/rosjava
- [ ] http://wiki.ros.org/rosjava
- [ ] http://wiki.ros.org/rosjava
- [ ] http://wiki.ros.org/rosjava
- [ ] http://wiki.ros.org/rosjava
[... 46 lines cut ...]
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/ariac/Tutorials/SystemSetup
- [ ] http://wiki.ros.org/robot_localization
- [ ] http://wiki.ros.org/robot_localization
- [ ] http://wiki.ros.org/robot_localization
- [ ] http://wiki.ros.org/robot_localization
- [ ] http://wiki.ros.org/robot_localization
- [ ] http://wiki.ros.org/robot_localization
- [ ] http://wiki.ros.org/robot_localization
- [ ] http://wiki.ros.org/rviz/Tutorials/Plugins(3a20)New(20)Tool(20)Type
- [ ] http://wiki.ros.org/rviz/Tutorials/Plugins(3a20)New(20)Display(20)Type
- [ ] http://wiki.ros.org/Robots/evarobot/tr
- [ ] http://wiki.ros.org/ariac/Tutorials/SensorInterface
- [ ] http://wiki.ros.org/ariac/Tutorials/SensorInterface

dirk-thomas, 2017-05-23:
The list needs to be deduplicated.

mikaelarguedas, 2017-05-31:
all these have been updated and the missing one have been ticketed and adressed : https://github.com/rosjava/android_core/issues/269.
Closing this

closed 2017-05-31 by mikaelarguedas