ros/rosdistro #13832: release new version of filters

nicolaerosia, 2017-02-08:
@tfoote @wjwwood I'm compiling ROS from source code using `rosinstall_generator` but I end up with an old version of filters which does not have the latest fixes (from 2014) and build breaks with testing disabled (`DCATKIN_ENABLE_TESTING=OFF`).

Could you please create a new tag and update the distro accordingly?

For reference, here is ros filters, https://github.com/ros/filters
Reproducing steps
```
# Generate a rosinstall file containing a list of repositories to packages
rosinstall_generator perception \
--rosdistro kinetic \
--deps \
--wet-only \
--upstream | tee perception.rosinstall

rosinstall_generator simulators \
--rosdistro kinetic \
--deps \
--wet-only \
--exclude \
stage_ros \
--upstream | tee simulators.rosinstall

rosinstall_generator viz \
--rosdistro kinetic \
--deps \
--wet-only \
--upstream | tee viz.rosinstall
```

wjwwood, 2017-02-08:
@tfoote will need to make a new version and release it, in which making a new flag and updating the distro is part of that process but not all of it.

nicolaerosia, 2017-02-08:
@wjwwood how are you guys using ROS, are you compiling it? Are you using rosinstall_generator ? With what flags?

wjwwood, 2017-02-08:
I use `rosinstall_generator` and build from source pretty often. I haven't run into this issue in particular in my development though.

Others use the binaries we provide for Ubuntu and just build the packages on top of that which they are working on actively.

tfoote, 2017-03-20:
https://github.com/ros/filters/commit/7664befa1319f306e3e2330e82fb99d817c727cd

https://github.com/ros/rosdistro/pull/14199
https://github.com/ros/rosdistro/pull/14197
https://github.com/ros/rosdistro/pull/14196

closed 2017-03-20 by tfoote