ros/rosdistro #14054: Unmet Dependencies when installing Kinetic on 16.04

superjax, 2017-03-01:
Hey there,

I've got a really weird problem trying to install ROS on a vanilla installation of Ubuntu 16.04.  When I install `ros-kinetic-desktop-full` apt-get only downloads something like 13kB worth of packages, and it appears that a bunch of the `python-ros*` packages are missing.   I've tried to install them all manually, but I am getting this output out of `aptitude` when trying to install `python-catkin-pkg-modules`.  

``` The following packages have unmet dependencies:
 python-catkin-pkg-modules : Conflicts: catkin but 0.6.16-4 is installed.
 python-catkin-pkg : Conflicts: catkin but 0.6.16-4 is installed.
```

I am unable to simultaneously install `python-catkin-pkg-modules`, `catkin`, `python-roslaunch` and `python-roslib`.  I'm not an `apt` wizard, but it seems to me that there is some version mismatching.

I have replicated this issue now on 6 computers, all with fresh 16.04 installations and different hardware.  Any ideas?

el3ment, 2017-03-01:
I am also having this issue.

dirk-thomas, 2017-03-01:
It looks like that you are mixing upstream Debian packages of ROS `catkin` with packages provided by the ROS repositories (`python-catkin-pkg-modules`). This is not a supported setup - see http://wiki.ros.org/UpstreamPackages for details.

You can either uninstall `catkin` and install `ros-kinetic-catkin` instead (using the packages provided by the official ROS apt repo) or remove the ROS repo from your apt list and use only Debian packages provided directly with Ubuntu.

el3ment, 2017-03-01:
How am I mixing them? I don't doubt that I am -- I just honestly don't know how it's happening.

dirk-thomas, 2017-03-01:
If you remove the ROS repository you can only install Debian packages available in Ubuntu. I am not sure but it looks like that there is no desktop-full package available. Also note that the upstream Debian packages are not prefixed with `ros-kinetic-` but mostly only with `ros-`. See https://wiki.debian.org/DebianScience/Robotics/ROS/Packages for a list of available packages.

el3ment, 2017-03-01:
If I'd like to install using the `package.ros.org` packages (I need the xacro package which hasn't made it into the Debian repository yet), how do I prevent apt from using the upstream packages when installing `ros-kinetic-desktop-full`?

dirk-thomas, 2017-03-01:
If you use the ROS apt repo and install `ros-kinetic-...` packages no packages from upstream will be installed. If you have some upstream packages installed apt should complain about it and you need to remove those.

el3ment, 2017-03-01:
After installing and trying to run `roscore` or `roslaunch` I find that neither were installed and that I am to install `python-roslaunch` which comes from `us.archive.ubuntu.com/ubuntu` not `packages.ros.org`. Attempting to install `python-roslaunch` gives an error that `python-roslib` has an issue with `catkin`. I confirmed that `ros-kinetic-python-catkin` is installed but `sudo apt install python-roslib` is looking for `catkin` not `ros-kinetic-catkin`.

dirk-thomas, 2017-03-01:
Please follow the [install instructions](http://wiki.ros.org/kinetic/Installation/Ubuntu) after uninstalling upstream Debian packages. `python-roslaunch` is an upstream Debian package. The equivalent in the ROS repo is called `ros-kinetic-roslaunch` (same for `roslib`).

superjax, 2017-03-01:
Okay, that makes sense.  I'm following the install instructions you have linked, but it seems that the real issue is that `ros-kinetic-desktop-full` isn't installing the proper packages, and I'm trying to install the upstream packages to fix it.  Any idea why `ros-kinetic-desktop-full` doesn't install `roscore` and `catkin` automatically?

el3ment, 2017-03-01:
I suppose the issue is that ros-kinetic-roslaunch was installed (via `sudo apt install ros-kinetic-ros*`), but it's not finding the command (and that the "help" command to install "python-roslaunch" is misleading). I was neglecting to source the `/opt/ros/kinetic/setup.bash` file.

superjax, 2017-03-01:
Aha!  That makes sense.  my automated scripts have the same issue.

Thanks!

closed 2017-03-01 by superjax

jediofgever, 2017-11-30:
This is my answer from stackoverflow.com  

After upgrading ubuntu 14.04 to 16.04 I run into the same issue. Adding following package sources to `/etc/apt/sources.list` worked for me. So just do `sudo gedit /etc/apt/sources.list` copy paste following lines. Apply steps from 1.2 to 1.7 in this link

```
deb http://mirrors.ustc.edu.cn/ubuntu/ xenial main restricted universe multiverse
deb http://mirrors.ustc.edu.cn/ubuntu/ xenial-security main restricted universe multiverse
deb http://mirrors.ustc.edu.cn/ubuntu/ xenial-updates main restricted universe multiverse
deb http://mirrors.ustc.edu.cn/ubuntu/ xenial-proposed main restricted universe multiverse
deb http://mirrors.ustc.edu.cn/ubuntu/ xenial-backports main restricted universe multiverse
deb-src http://mirrors.ustc.edu.cn/ubuntu/ xenial main restricted universe multiverse
deb-src http://mirrors.ustc.edu.cn/ubuntu/ xenial-security main restricted universe multiverse
deb-src http://mirrors.ustc.edu.cn/ubuntu/ xenial-updates main restricted universe multiverse
deb-src http://mirrors.ustc.edu.cn/ubuntu/ xenial-proposed main restricted universe multiverse
deb-src http://mirrors.ustc.edu.cn/ubuntu/ xenial-backports main restricted universe multiverse

```

tfoote, 2017-11-30:
@jediofgever The apt sources are unrelated to this issue.

rayxyz, 2017-12-01:
Hi, guys, I've solved the same problem by adding deb http://cz.archive.ubuntu.com/ubuntu xenial main universe to the file /etc/apt/sources.list, and run the command sudo apt-get update, then install the kinetic by run sudo apt-get install ros-kinetic-desktop-full , everything went smoothly. You guys can visit the link https://packages.ubuntu.com/xenial/al... for more detail. The problem is the lack of python-qt-binding in the system or something goes with the packages.

tfoote, 2017-12-01:
@rayxyz That's also a different issue.

aravindharinair, 2017-12-12:
Hey,
I'm new to ROS and I don't have much experience working with Linux but I wanted to learn ROS for a college project. Now, following Jan Bernlöhr's guide ([here](https://janbernloehr.de/2017/06/10/ros-windows)), I installed Ubuntu on the Windows Subsystem but when running `roscore`. it's asking me to install `python ros-launch` which, when I try to install shows:
`The following packages have unmet dependencies:`
`python-roslaunch : Depends: python-roslib but it is not going to be installed`
And it goes back until `catkin`.
Now, as per what @el3ment said, I found that `ros-lunar-roslaunch` is installed but I have no idea how to proceed from there. I'm tried what @rayxyz suggested but it's showing:
`ros-lunar-desktop-full is already the newest version (1.3.1-0xenial-20171120-113826-0800).`
`0 upgraded, 0 newly installed, 0 to remove and 127 not upgraded.`
Any help regarding this would be appreciated.
Thanks in advance !

tfoote, 2017-12-12:
@aravindharinair 

In general for help please ask on https://answers.ros.org following our [Support Guidelines](http://wiki.ros.org/Support)

At the moment it's down [due to infrastructure maintenance](https://status.ros.org/incidents/jfql57hvr474) on our provider. But you need to source /opt/ros/lunar/setup.bash before running anything to get the lunar versions of things instead of the [upstream version](http://wiki.ros.org/UpstreamPackages).

aravindharinair, 2017-12-12:
Ok. Will use that for any further queries. Thank you for letting me know.

I had forgotten to source it. My bad. Sorry for the inconvenience!
Thank you!

pianoslum, 2018-06-27:
I'm sorry, but I have the same problem and couldn't figure out how to solve it.
I'm on Debian 9 and followed the guide http://wiki.ros.org/melodic/Installation/Debian . However, when I try to install ros-perception, I get the following error:
 python-catkin-pkg : Kollidiert mit: catkin aber 0.7.4-4 soll installiert werden
 python-catkin-pkg-modules : Kollidiert mit: catkin aber 0.7.4-4 soll installiert werden
 (translation : collides with: catkin but 0.7.4-4 should be installed)

I don't know how I could have mixed upstream or ros-packages - can you give me a hint?
Thanks!

tfoote, 2018-06-27:
@pianoslum 
In general for help please ask on https://answers.ros.org following our Support Guidelines

The package `ros-perception` is from Debian directly and is not melodic. Melodic would be `ros-melodic-perception`  Please review http://wiki.ros.org/UpstreamPackages

lentilwallop, 2019-03-13:
vXCVCVC

lentilwallop, 2019-03-13:
pass the buck...   asses like tfoote just pass the buck and never help anything

roobooot, 2020-03-08:
sudo dpkg -r --force-all python-rospkg
Afterwards, sudo apt --fix-broken install If you wanna reinstall ros, run sudo apt-get remove ros-* to remove ros. Then, install ROS from https://wiki.ros.org/ROS/Installation

refer to https://askubuntu.com/a/1215647/1051598

roobooot, 2020-03-08:
`sudo dpkg -r --force-all python-rospkg`
Afterwards, `sudo apt --fix-broken install `If you wanna reinstall ros, run `sudo apt-get remove ros-*` to remove ros. Then, install ROS from https://wiki.ros.org/ROS/Installation

AyanHBK123, 2021-01-20:
I am trying to install ROS kinetic on Debain Buster but I am getting this error; 
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 python-rosdep-modules : Depends: python-catkin-pkg-modules (>= 0.4.0) but it is not installed
                         Depends: python-rosdistro-modules (>= 0.7.5) but it is not installed
 python-rospkg-modules : Depends: python-catkin-pkg-modules but it is not installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).

I tried to fix the issue 
from here https://github.com/ros/rosdistro/issues/14054#issuecomment-596168985

but now i am getting this error:
Can you help how to solve this issue ??

clalancette, 2021-01-20:
Please ask questions like this on https://answers.ros.org .  That is the central, searchable place to help for ROS.

mxshen, 2023-12-29:
> `sudo dpkg -r --force-all python-rospkg` Afterwards, `sudo apt --fix-broken install `If you wanna reinstall ros, run `sudo apt-get remove ros-*` to remove ros. Then, install ROS from https://wiki.ros.org/ROS/Installation

Thx bro! I tried many methods, but only yours worked.