ros/rosdistro #14112: Lunar Release Checklist

mikaelarguedas, 2017-03-06:
This ticket is used to collect things that need to be done for the upcoming Lunar release. (copied from the [Kinetic Release ticket](https://github.com/ros/rosdistro/issues/9944))

Existing Task Sets:
- [ ] Lunar kickoff list https://github.com/ros/rosdistro/issues/13550
- [ ] Lunar documentation audit
- [ ] Lunar community documentation audit

---
Documentation:
- [ ] Update verbiage on these wiki pages, e.g. 'upcoming release' vs. 'current release'
  - [ ] http://wiki.ros.org/lunar/Installation
  - [ ] http://wiki.ros.org/lunar
- [ ] Turtlesim icon
- [ ] M-turtle name

---
ROS Wiki
- [ ] make distroN the default distro
- [ ] Version macro for distroN needs to be updated from “Expected in distroN” to “distroN”
- [ ] Manually redeploy wiki once finished by pulling repo and restarting Apache instance
- [ ] Update symlink for default API doc:
  - /home/rosbot/docs/api → /home/rosbot/docs/distroN/api
  - /home/rosbot/latest → /home/rosbot/docs/distroN
- [ ] Update http://wiki.ros.org/Distributions
  - reflect released status (update from pending)
- [ ] Update http://wiki.ros.org/ROS/Installation to point to distroN
- [ ] remove EOL'd from header doc links

---
ROS browse
- [ ] update default distro

---
Communication
- [ ] Announcement email to ros-users pointing to migration guide 
  - Post to ros-users
  - Post to blog
  - Post to G+
  - Post to Twitter

---
www.ros.org
- [ ] Update www.ros.org bottom row widget to point to new release with logo
- [ ] Update www.ros.org/install to point to new release with logo

dirk-thomas, 2017-05-24:
@mikaelarguedas The table in [section 5](http://wiki.ros.org/Distributions#Which_distribution_to_use) needs an update too.

mikaelarguedas, 2017-05-24:
I didn't update it because I thought we did not officially encourage people to switch until more packages (like navigation and moveit) were released. I'm happy to update it if it's not the case

mikaelarguedas, 2017-05-31:
Lunar Loggerhead has been released :tada: and the community documentation will be improved along packages being released. Closing this

closed 2017-05-31 by mikaelarguedas