ros/rosdistro #14360: kinetic job reconfiguration is failing

hrnr, 2017-03-28:
Hi,

kinetic jobs reconfiguration task `Krel_reconfigure-jobs` is failing since Mar 23 due to circular dependecies between `roch_bringup, roch_capabilities`. This was introduced by #14299. This prevents releasing new packages for kinetic.

Is anybody aware of it?

Cheers,
Jiri

dirk-thomas, 2017-03-28:
@tfoote is in charge of Kinetic and has received notification email for all of the failing jobs. Please also improve the unit tests in this repo to catch the problem before merging in the future.

dirk-thomas, 2017-03-28:
The same is obviously the case for Indigo.

tfoote, 2017-03-29:
Also at issue:
Indigo: https://github.com/ros/rosdistro/pull/14268
Debinc: https://github.com/ros/rosdistro/pull/14306

tfoote, 2017-03-29:
Kinetic reconfiguring is running, and Indigo is waiting for kinetic to finish to run. I think it should work after kinetic finishes. I'll check in the morning.

tfoote, 2017-03-29:
Indigo and Kinetic are both reconfiguring successfully now.

closed 2017-03-29 by tfoote