ros/rosdistro #14586: local prerelease test fails on running locale-gen

bmagyar, 2017-04-21:
Have I missed something obvious? 
A few days ago when I was trying to run a prerelease script I got the error below:

```
/bin/sh: 1: locale-gen: not found
Removing intermediate container 888ff79a49a0
The command '/bin/sh -c locale-gen en_US.UTF-8' returned a non-zero code: 127
```

mikaelarguedas, 2017-04-21:
We had a 24hour window (9 days ago) when locales were not available in the
generated dockerfiles, leading to gen-locales invocation failure (
https://github.com/ros-infrastructure/ros_buildfarm/pull/415).

That is very likely the reason. Can you still reproduce the error? If so
let us know I may have missed a place to install locales.

@dirk-thomas is a new release of ros_buildfarm with the fix a sensible
thing to do?

dirk-thomas, 2017-04-21:
@mikaelarguedas Currently we shouldn't make a `ros_buildfarm` release due to other problems being investigated.

@bmagyar You can patch your local installation by updating the files as mentioned in the referenced PR. You will find the installed templates in `/usr/lib/python2.7/dist-packages/ros_buildfarm/templates` / `/usr/lib/python3/dist-packages/ros_buildfarm/templates` depending on your Python version. It think it should be sufficient to only patch the template `devel/devel_task.Dockerfile.em`.

I will close this since the problem has already been fixed. Please feel free to continue commenting on this ticket.

closed 2017-04-21 by dirk-thomas

dirk-thomas, 2017-04-21:
The just released new version of `ros_buildfarm` 1.3.1 should work out-of-the-box.

bmagyar, 2017-04-22:
Thanks for the quick fix! It works ;)