ros/rosdistro #15082: Missing Kinetic dependencies for Gentoo Linux

allenh1, 2017-05-26:
Here's the list from my script:

```
>>>> Unresolved:
>>>> ethercat_grant:
>>>>   ['libcap-dev']
>>>> homer_mary_tts:
>>>>   ['python-pyaudio']
>>>> async_web_server_cpp:
>>>>   ['python-websocket']
>>>> rtt:
>>>>   ['omniorb']
>>>> parrot_arsdk:
>>>>   ['yasm', 'libavahi-core-dev', 'libavahi-client-dev']
>>>> jsk_perception:
>>>>   ['python-h5py', 'python-dlib', 'python-chainer-pip', 'leveldb', 'python-sklearn', 'python-fcn-pip']
>>>> roch_ftdi:
>>>>   ['ftdi-eeprom', 'libusb-dev']
>>>> swri_transform_util:
>>>>   ['libgeos++-dev']
>>>> ecto:
>>>>   ['xdot']
>>>> world_canvas_client_py:
>>>>   ['python-pymongo']
>>>> ocl:
>>>>   ['netcdf', 'lua-dev']
>>>> pose_cov_ops:
>>>>   ['mrpt']
>>>> libsensors_monitor:
>>>>   ['libsensors4-dev']
>>>> ps3joy:
>>>>   ['libusb-dev']
>>>> openrtm_aist:
>>>>   ['omniorb']
>>>> thormang3_opc:
>>>>   ['thormang3_action_script_player', 'thormang3_foot_step_generator', 'thormang3_offset_tuner_client']
>>>> assimp_devel:
>>>>   ['openssl', 'ca-certificates']
>>>> mrpt_rawlog:
>>>>   ['mrpt']
>>>> rosbridge_server:
>>>>   ['python-twisted-core']
>>>> rtabmap:
>>>>   ['libfreenect-dev']
>>>> rosdoc_lite:
>>>>   ['python-kitchen']
>>>> rosserial_python:
>>>>   ['python-serial']
>>>> mrpt_rbpf_slam:
>>>>   ['mrpt']
>>>> node_manager_fkie:
>>>>   ['xterm', 'python-docutils']
>>>> downward:
>>>>   ['gawk', 'time', 'g++-static', 'ca-certificates']
>>>> homer_map_manager:
>>>>   ['sdl']
>>>> people:
>>>>   ['people_velocity_tracker', 'leg_detector']
>>>> libphidget21:
>>>>   ['libusb-dev']
>>>> feed_the_troll:
>>>>   ['python-termcolor']
>>>> nmea_navsat_driver:
>>>>   ['python-serial']
>>>> webkit_dependency:
>>>>   ['python-qt5-bindings-webkit']
>>>> roomblock_bringup:
>>>>   ['python-serial']
>>>> joy_mouse:
>>>>   ['python-pyudev']
>>>> mrpt_ekf_slam_3d:
>>>>   ['mrpt']
>>>> spacenav_node:
>>>>   ['spacenavd']
>>>> homer_robot_face:
>>>>   ['libpulse-dev', 'libesd0-dev', 'libestools-dev', 'libqt5x11extras5-dev', 'festival-dev']
>>>> rqt_py_trees:
>>>>   ['python-termcolor']
>>>> libcmt:
>>>>   ['ca-certificates', 'openssl']
>>>> pyros:
>>>>   ['python-six']
>>>> jsk_tools:
>>>>   ['python-texttable', 'python-colorama', 'python-requests', 'python-percol', 'python-pygithub3', 'python-slacker-cli']
>>>> pyros_config:
>>>>   ['python-pytest', 'python-six']
>>>> rocon_gateway:
>>>>   ['python-crypto']
>>>> jsk_teleop_joy:
>>>>   ['python-pygame']
>>>> pddl_planner:
>>>>   ['time']
>>>> rqt_multiplot:
>>>>   ['libqwt6']
>>>> nav2d_karto:
>>>>   ['suitesparse']
>>>> master_discovery_fkie:
>>>>   ['python-avahi']
>>>> launch_tools:
>>>>   ['xdot']
>>>> jsk_data:
>>>>   ['python-gdown-pip', 'python-click']
>>>> audio_play:
>>>>   ['gstreamer1.0-plugins-base', 'gstreamer1.0-plugins-ugly', 'gstreamer1.0-plugins-good', 'gstreamer1.0', 'libgstreamer-plugins-base1.0-dev', 'libgstreamer1.0-dev']
>>>> mrpt_bridge:
>>>>   ['mrpt']
>>>> openhrp3:
>>>>   ['python-omniorb']
>>>> kobuki_qtestsuite:
>>>>   ['pyqt5-dev-tools']
>>>> mrpt_map:
>>>>   ['mrpt']
>>>> multisense_ros:
>>>>   ['libturbojpeg']
>>>> mvsim:
>>>>   ['mrpt']
>>>> mrpt_localization:
>>>>   ['mrpt']
>>>> message_relay:
>>>>   ['python-cheetah']
>>>> world_canvas_server:
>>>>   ['python-pymongo']
>>>> ardrone_autonomy:
>>>>   ['sdl']
>>>> sparse_bundle_adjustment:
>>>>   ['libblas-dev', 'suitesparse']
>>>> audio_capture:
>>>>   ['gstreamer1.0-plugins-base', 'gstreamer1.0-plugins-ugly', 'gstreamer1.0-plugins-good', 'gstreamer1.0', 'libgstreamer-plugins-base1.0-dev', 'libgstreamer1.0-dev']
>>>> py_trees_ros:
>>>>   ['python-termcolor']
>>>> nerian_sp1:
>>>>   ['sdl']
>>>> jsk_recognition_utils:
>>>>   ['python-skimage', 'cython']
>>>> jsk_pcl_ros_utils:
>>>>   ['python-sklearn']
>>>> korg_nanokontrol:
>>>>   ['python-pygame']
>>>> uwsim:
>>>>   ['libxml++-2.6']
>>>> marshmallow:
>>>>   ['python-dateutil', 'python-simplejson']
>>>> dynamixel_driver:
>>>>   ['python-serial']
>>>> visp_auto_tracker:
>>>>   ['libdmtx-dev', 'zbar']
>>>> sound_play:
>>>>   ['gstreamer1.0-plugins-base', 'gstreamer1.0', 'gstreamer1.0-plugins-ugly', 'python-gi', 'gstreamer1.0-plugins-good']
>>>> pyzmp:
>>>>   ['python-zmq']
>>>> jsk_pcl_ros:
>>>>   ['python-sklearn']
>>>> log_server:
>>>>   ['python-expiringdict']
>>>> openrtm_aist_python:
>>>>   ['python-omniorb']
>>>> libg2o:
>>>>   ['suitesparse']
>>>> ffha:
>>>>   ['gawk']
>>>> rospilot:
>>>>   ['python-mapnik', 'python-cherrypy', 'python-tilestache', 'python-colorama', 'osm2pgsql', 'postgresql-9.x-postgis', 'mapnik-utils', 'python-serial', 'libturbojpeg', 'libmicrohttpd']
>>>> rocon_hub:
>>>>   ['avahi-utils', 'redis-server']
>>>> mrpt_ekf_slam_2d:
>>>>   ['mrpt']
>>>> rtctree:
>>>>   ['python-omniorb']
>>>> plotjuggler:
>>>>   ['libqt5-svg-dev']
>>>> ros_peerjs:
>>>>   ['libxss1', 'nodejs', 'xvfb', 'libgconf2', 'libnss3-dev', 'libxtst-dev', 'libasound2-dev', 'npm', 'nodejs-legacy']
>>>> hrpsys:
>>>>   ['python-tk', 'sdl']
>>>> slic:
>>>>   ['ca-certificates', 'openssl']
>>>> rtshell:
>>>>   ['python-omniorb']
>>>> openni2_camera:
>>>>   ['libopenni2-dev']
>>>> kobuki_ftdi:
>>>>   ['ftdi-eeprom', 'libusb-dev']
>>>> librealsense:
>>>>   ['dkms']
>>>> trac_ik_lib:
>>>>   ['libnlopt-dev', 'libnlopt0']
>>>> zbar_ros:
>>>>   ['zbar']
>>>> rosserial_xbee:
>>>>   ['python-serial']
>>>> webtest:
>>>>   ['python-bs4', 'python-waitress', 'python-six', 'python-webob']
>>>> object_recognition_reconstruction:
>>>>   ['meshlab']
>>>> catkin_pip:
>>>>   ['python-pip']
>>>> swri_geometry_util:
>>>>   ['libgeos++-dev']
>>>> mrpt_icp_slam_2d:
>>>>   ['mrpt']
>>>> visp:
>>>>   ['libfreenect-dev', 'libusb-dev', 'libdmtx-dev', 'libcoin80-dev', 'libdc1394-dev', 'zbar']
>>>> mrpt_graphslam_2d:
>>>>   ['mrpt']
>>>> rosbridge_library:
>>>>   ['python-bson']
>>>> ff:
>>>>   ['openssl', 'ca-certificates']
>>>> tile_map:
>>>>   ['libjsoncpp']
>>>> euslisp:
>>>>   ['xfonts-100dpi', 'xfonts-75dpi']
>>>> zeroconf_avahi:
>>>>   ['libavahi-core-dev', 'libavahi-client-dev']
>>>> py_trees:
>>>>   ['python-enum34']
>>>> rtsprofile:
>>>>   ['python-omniorb']
>>>> razor_imu_9dof:
>>>>   ['python-serial']
```

allenh1, 2017-05-26:
@Alessandro-Barbieri do you want to work on this?

mikaelarguedas, 2017-05-26:
is it possible to generate a list of the missing keys for all ROS distros and removing duplicates? that would be easier to process IMO

allenh1, 2017-05-26:
Sure. One moment.

Alessandro-Barbieri, 2017-05-26:
@allenh1 yess

allenh1, 2017-05-26:
Updating the original list.

Alessandro-Barbieri, 2017-05-26:
I see `openssl`, there is a problem: how to implement this?
```
ssl? (
    !libressl? ( dev-libs/openssl:0= )
    libressl? ( dev-libs/libressl:0= )
)
```

see https://wiki.gentoo.org/wiki/Project:LibreSSL for reference

Alessandro-Barbieri, 2017-05-26:
If you merge my pull request some of them will go away or will be done all in one commit?

Alessandro-Barbieri, 2017-05-26:
Another question: there is a way to practically edit that huge `yaml` files?

allenh1, 2017-05-26:
You certainly need not remove them all in one commit. I'll update the list as we go. I removed a whole lot of packages that didn't need to be in the list already. If you want to hold off, I'll modify the thing generating the ebuilds so that it gives one list at the very end for all relevant distros. The current list is a little daunting.

allenh1, 2017-05-26:
For the `openssl` issue, that's a serious problem... Hm... Maybe for now we just put `dev-libs/openssl` until I come up with a better way of handling this. 

Perhaps it would be good to write a small `virtual/ssl` ebuild to put in the repo that installs one of those, as conditioned there.

@mikaelarguedas  thoughts?

allenh1, 2017-05-26:
@Alessandro-Barbieri I'm working on shortening that list for us, don't worry. I'm generating just what would be needed to do indigo, kinetic and lunar. It'll be about half an hour to process this.

Alessandro-Barbieri, 2017-05-28:
Python and ruby ready to be merged.

allenh1, 2017-05-29:
@Alessandro-Barbieri Updated the list

allenh1, 2017-05-29:
List now only contains those necessary for lunar & kinetic to have all deps satisfied.

allenh1, 2017-05-29:
Updated to add the deps for indigo. The list is now complete.

allenh1, 2017-05-29:
@Alessandro-Barbieri please reference this issue in any related PR's.

Alessandro-Barbieri, 2017-05-29:
What to do with the `*-pip` packages?

allenh1, 2017-05-29:
Pip is not supported anymore, so I don't think they need to be implemented. Just the list above is necessary.

Alessandro-Barbieri, 2017-05-30:
For the openssl thing, the libressl overlay provide a fake `openssl` package that is co-installable with `libressl` so no need for a  virtual an we'll use `openssl`.

mikaelarguedas, 2017-05-30:
> What to do with the *-pip packages?

pip is supported for people building their packages from source and using rosdep to install their dependencies. But is not used by the buildfarm, meaning that every **released** package doesn't use any `-pip` key. So I think it's safe to not add them for now and keep this list more digestible

Alessandro-Barbieri, 2017-05-31:
Finished, time to check missing packages @allenh1

Alessandro-Barbieri, 2017-06-03:
Do you update the list manually or with a script?

allenh1, 2017-06-03:
I update them manually

allenh1, 2017-06-03:
But it's probably easier to just replace this list with an updated one when the changes are merged.

Alessandro-Barbieri, 2017-06-08:
Can you rebuild the list?

allenh1, 2017-06-08:
Yes, kinetic is rebuilding atm.

allenh1, 2017-06-11:
> Can you rebuild the list?

Done.

mikaelarguedas, 2017-06-11:
Note: There seem to be some ROS packages and some pip keys in the new list that can be ignored

Alessandro-Barbieri, 2017-06-11:
Ignoring python for now

    `disper` in overlay https://bugs.gentoo.org/show_bug.cgi?id=281024
    `dkms` in overlay https://bugs.gentoo.org/show_bug.cgi?id=100754
    `g++-static` `sys-devel/gcc`
    `gstreamer0.10-pocketsphinx` `app-accessibility/pocketsphinx`
    `language-pack-de` ubuntu only package
    `language-pack-en` ubuntu only package
    leg_detector
    libcap-dev
    libcoin80-dev
    libesd0-dev
    libfreenect-dev
    mapnik-utils
    mrpt
    nodejs-legacy
    npm
    opende
    people_velocity_tracker
    qtmobility-dev
    sr_moveit_config
    thormang3_action_script_player
    thormang3_foot_step_generator
    thormang3_offset_tuner_client
    xfonts-100dpi
    xfonts-75dpi
    xvfb

mikaelarguedas, 2017-06-12:
All the "what is this?" are ROS packages and thus can be removed from the list

leio, 2017-06-12:
xfonts-100dpi is probably media-fonts/font-bitstream-100dpi
xfonts-75dpi is probably font-bitstream-75dpi
But we install in /usr/share/fonts prefix, not /usr/share/fonts/X11 like debian; and I don't see any ISO-8859-* files like debian filelist lists

Alessandro-Barbieri, 2017-06-12:
> and I don't see any ISO-8859-* files like debian filelist lists

These are in the `media-fonts/font-adobe-100dpi` and `media-fonts/font-adobe-75dpi`

Alessandro-Barbieri, 2017-06-12:
@leio thanks

Alessandro-Barbieri, 2017-07-04:
How can I know which ros packages pulls in these dependencies?

allenh1, 2017-07-04:
@Alessandro-Barbieri You can run my script and watch the output of it (it shows up in red when things fail to resolve).

tfoote, 2017-08-03:
@allenh1 Can we close this now that ros-overlay is generally running?

allenh1, 2017-08-03:
@tfoote sure. That's reasonable.

closed 2017-08-03 by allenh1