ros/rosdistro #15088: No definition of [python-qt5-bindings-webkit] for OS

Alessandro-Barbieri, 2017-05-27:
Running `sudo python3 /usr/bin/rosdep install --verbose --from-paths src --ignore-src --rosdistro jade -y` gives me this error

```
resolution: portage [dev-python/netifaces, media-libs/tiff, dev-util/cppunit, dev-qt/qtcore, media-libs/libv4l, dev-cpp/eigen, dev-python/nose, dev-python/pygraphviz, dev-qt/qtgui, media-libs/collada-dom, app-arch/lz4, dev-util/rosdep, dev-libs/protobuf, media-gfx/graphviz, media-libs/libogg, dev-libs/boost, media-video/ffmpeg, dev-games/ogre, sci-libs/proj, dev-python/psutil, dev-libs/console_bridge, dev-cpp/gtest, dev-qt/qtwidgets, dev-python/catkin_pkg, media-libs/assimp, sys-libs/zlib, python, sci-libs/pcl, dev-python/numpy, =x11-libs/fltk-1*, dev-python/pycairo, dev-qt/qtopengl, pyyaml, dev-libs/poco, virtual/pkgconfig, media-libs/libjpeg-turbo, dev-python/PyQt5, dev-python/pyopengl, dev-lisp/sbcl, x11-themes/tango-icon-theme, dev-util/cmake, dev-python/mock, sci-libs/vtk, sci-electronics/gazebo, dev-python/defusedxml, dev-cpp/yaml-cpp, dev-python/rospkg, dev-libs/apr, dev-libs/apr-util, dev-python/paramiko, dev-libs/urdfdom, dev-python/pillow, dev-python/matplotlib, media-libs/libtheora, media-libs/mesa, media-libs/jasper, x11-libs/gtk+, net-misc/curl, sys-devel/libtool, util-linux, dev-libs/tinyxml, dev-libs/urdfdom_headers, media-libs/qhull, app-arch/bzip2, dev-python/sip, dev-libs/log4cxx, media-gfx/pydot, media-libs/libpng, dev-python/empy, app-admin/hddtemp, dev-python/coverage]
uninstalled: [sci-libs/vtk, sci-libs/proj, x11-themes/tango-icon-theme, sci-libs/pcl, dev-python/mock, media-libs/collada-dom, sci-electronics/gazebo, dev-lisp/sbcl, dev-cpp/gtest]
uninstalled dependencies are: [sci-libs/vtk, sci-libs/proj, x11-themes/tango-icon-theme, sci-libs/pcl, dev-python/mock, media-libs/collada-dom, sci-electronics/gazebo, dev-lisp/sbcl, dev-cpp/gtest]
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
webkit_dependency: No definition of [python-qt5-bindings-webkit] for OS [gentoo]
```

but https://github.com/ros/rosdistro/blob/master/rosdep/python.yaml#L2163 is correct and I already have `dev-python/PyQt5` installed.

I found others with the same issue here:
http://answers.ros.org/question/262190/webkit_dependency-no-definition-of-python-qt5-bindings-webkit-for-os-osx/
https://github.com/ros/rosdistro/pull/15086

mikaelarguedas, 2017-05-27:
Have your tried running `rosdep update` ? this rule has been added yesterday and may not be in your local rodep database yet.

It's surprising that any key resolution for jade gives `python-qt5-bindings-webkit` given that jade targets qt4 and not qt5.

The two referenced issues are unrelated because they both tried to use a rule that is **not** in the rosdep database (for various reasons)

Alessandro-Barbieri, 2017-05-27:
I run `rosdep update` multiple times yesterday and today

Alessandro-Barbieri, 2017-05-27:
Same with `kinetic`

Alessandro-Barbieri, 2017-05-28:
Ignore the `jade` thing, it was a mistake (run with `jade` instead of `kinetic`)

allenh1, 2017-05-29:
This is probably the result of rosdep using a cache. I think it should be good if you try it now.

Alessandro-Barbieri, 2017-05-29:
`rosdep` cache is in the `.ros` folder or server side? I'll retry deleting it

allenh1, 2017-05-29:
If you do a `rosdep update` you shouldn't have to delete anything at this point.

Alessandro-Barbieri, 2017-05-29:
Deleted the folder and run `rosdep update`, same error ...

allenh1, 2017-05-29:
Strange... @mikaelarguedas thoughts?

Alessandro-Barbieri, 2017-05-29:
I run it in verbose mode but isn't so verbose

Alessandro-Barbieri, 2017-05-29:
Found the arcane: don't run it with `sudo`

allenh1, 2017-05-29:
Ok, so it is ok?

allenh1, 2017-05-30:
@Alessandro-Barbieri you're not supposed to run `rosdep udpate` with sudo, so I think this can be closed.

Alessandro-Barbieri, 2017-05-30:
I run `rosdep update` as normal user and `rosdep install` with sudo

allenh1, 2017-05-30:
Ok, I see what you're saying. So you did

```
rosdep update
sudo rosdep install --from-paths=...
```
right? 

Do you know if this works?
```
# rosdep update
# rosdep install --from-paths=~user/...
```

Alessandro-Barbieri, 2017-05-30:
works

allenh1, 2017-05-30:
I figured that would be the case. Thanks for the info.

serser, 2017-05-31:
I am on Mac OS Sierra with the same issue, provided solutions don't work, it seems it doesn't support `python-qt5-bindings-webkit` on Mac OS according to `python.yaml` file, can someone give me a hint?

mikaelarguedas, 2017-05-31:
Indeed there is no rule for this package for Mac OS in the rosdep database. If there is an equivalent package in homebrew please submit a PR to https://github.com/ros/rosdistro/blob/master/rosdep/osx-homebrew.yaml to add it (contributing guide available [here](http://docs.ros.org/independent/api/rosdep/html/contributing_rules.html))

GoodBoy962, 2017-06-22:
@allenh1 I tried to run this : 

**rosdep update
sudo rosdep install --from-paths=..**

but it didn't help me

tfoote, 2019-01-23:
Since this is 18 months old. Closing.

closed 2019-01-23 by tfoote