ros/rosdistro #15164: release python packages in artful

mikaelarguedas, 2017-06-06:
@clalancette FYI

---
- [x] Add newly supported platforms to list of platforms for each Python package (in `stdeb.cfg`), in order of dependency (for ROS M, do this as soon as debs for new Ubuntu are available, in pre-alpha!):
  - [x] rospkg (uploaded, no need to rerelease): https://github.com/ros-infrastructure/rospkg/commit/9c17190a8ef8e4428ef58c4038efd86c0e404fa4
  - [x] catkin_pkg (uploaded, no need to rerelease): https://github.com/ros-infrastructure/catkin_pkg/commit/024ea337447aa5d0352437c5ed95c0bdb7c39186
  - [x] rosdistro (uploaded, no need to rerelease): https://github.com/ros-infrastructure/rosdistro/commit/bde376de0b508fae22f034a5a6fbe72980a92aa6
  - [x] rosdep (uploaded, no need to rerelease):  https://github.com/ros-infrastructure/rosdep/commit/b1de10cf9691371f2854fffb8d239fe6f1f37c2c
  - [x] vcstools (uploaded, no need to rerelease): https://github.com/vcstools/vcstools/commit/ada7e69273bee5138e40df29065d7aa6be663ab9
  - [x] wstool (uploaded, no need to rerelease): https://github.com/vcstools/wstool/commit/c793ad8b7408b3005edc7f7ef6f2b855442e766c
  - [x] rosinstall (uploaded, no need to rerelease): https://github.com/vcstools/rosinstall/commit/47d7f64370fbe99469cfbe0bb332cec500a0ef65
  - [x] bloom (uploaded, no need to rerelease):  https://github.com/ros-infrastructure/bloom/commit/eddcc3573f28da0fa058a5162c458e437d2c5d94
  - [x] rosinstall_generator (uploaded, no need to rerelease):  https://github.com/ros-infrastructure/rosinstall_generator/commit/577258363f1511f5d541e3faf98a751e0b9453ee
  - [x] catkin_tools (uploaded, no need to rerelease): https://github.com/catkin/catkin_tools/commit/ba65534e43ab5bab65d604bd6795901e59bce7d4
  - [ ] vcstool: @dirk-thomas do you yant me to PR this or do you prefer to commit the changes directly ?
- [x] Use [`ros_release_python`](https://github.com/ros-infrastructure/ros_release_python) to make the currently released versions available in the new distro. Use the `--include` option (done for all except vcstool)

dirk-thomas, 2017-06-06:
The current releases are incomplete. E.g. `python-rosdistro-modules` is not in the apt repo for `artful`. I assume that the `--include` command doesn't consider the additional `-modules` packages atm.

dirk-thomas, 2017-06-06:
I updated `vcstool` and ran `--include` for it.

dirk-thomas, 2017-06-06:
I updated `ros_buidlfarm` (https://github.com/ros-infrastructure/ros_buildfarm/commit/ced64b5d636323f0dd9d111b02cd986c554ff9fe). I didn't run `--include` for it yet since it requires the same fix as the other missing packages.

dirk-thomas, 2017-07-13:
I fixed the problem in ros-infrastructure/ros_release_python#19 and ran `--include` again for the Python packages with more than one section in their `stdeb.cfg` files.

closed 2017-07-13 by dirk-thomas

dirk-thomas, 2017-07-13:
While all our packages are now available (at least in repositories.ros.org) they are not being installed on `artful` since new Debian packages have been imported which have a debinc of `2` and therefore a higher version number than our packages. While in general it would be "ok" if they ship the same version their Debian packages are not side-by-side installable for Python 2 and Python 3 since they don't split out the modules from the conflicting binaries.

dirk-thomas, 2017-07-13:
@jspricke FYI the problem in `artful` regarding the upstream Debian packages. Do you have a proposal how to address this? Maybe it would be best if the Debian packages are also being split in the same way to allow side-by-side installation.

jspricke, 2017-07-13:
I'm working on integrating Python 3 support for all packages that are
in Debian, but I cannot guarantee that this will be finished in time, as
they need to go through the Debian new queue (managed by Debian ftp
masters) and import freeze for Artful is on August 24th.

Cheers Jochen

dirk-thomas, 2017-07-13:
What do you suggest as a fallback if they don't make it in?

jspricke, 2017-07-13:
Increase the Debian revision number on your side?

jspricke, 2017-07-13:
An other (probably better) option would be to integrate the Debian packages later.

dirk-thomas, 2017-07-13:
Increasing the debinc would only work until updated also bumps the debinc. Therefore I don't think that is a long term solution.

With the need to be able to release changes in a timely manner I don't think using upstream is realistic in the near term future. The constraints for releasing new versions into Debian are sometimes too restrictive for the ROS community.

I guess we might have to bump the patch version of all packages affected. That will make sure to work at least until the upstream Debian packages catch up with that (hopefully only on the next Debian distro).

jspricke, 2017-07-13:
I would upload new patch versions to Debian, so you would gain nothing.

dirk-thomas, 2017-07-13:
Then we need a sustainable way to ensure that the ROS packages are being used. Not sure what that would be other than maybe using debinc 100 for our packages...

jspricke, 2017-07-13:
Given that you are using random Debian increments already (https://github.com/ros-infrastructure/ros_buildfarm/pull/460) I think it would work as an interim solution.

dirk-thomas, 2017-07-13:
The `ros_buildfarm` is only used for ROS packages. The Python packages are handled by a different release toolchain: https://github.com/ros-infrastructure/ros_release_python which is based on `stdeb`. I still have to check if `stdeb` allows us to specify a debinc - currently be only use the default.

dirk-thomas, 2017-07-14:
Thank you for the feedback. The next releases of `catkin_pkg`, `rosdistro`, `rospkg`, and `ros_buildfarm` will use a debinc of 100 for the non-modules packages.

jspricke, 2017-07-14:
ros_buildfarm is not in Debian and catkin_pkg just git accepted, so you
can leave both of them out.