ros/rosdistro #15210: outdated dependencies and ROS

Alessandro-Barbieri, 2017-06-12:
ROS still depend on outdated packages like `gstreamer0.10` and `qt4` and them will be removed from Gentoo (gstreamer0.10 is security vulnerable).
There is a policy that prevents user on depending on them?
There is a milestone in removing these packages completely from other distros too?

mikaelarguedas, 2017-06-12:
Qt4 is the targeted qt major for indigo and Jade so it will stay a required dependency for packages released on these distributions until they both go EOL. Event after that we will keep them in the rosdep database for users using EOL distros.
Thus there won't be (AFAIK) any policy preventing people from depending on them, and they will not be removed from the database for the older distros that target them.
Newer distros will target new versions (e.g kinetic and lunar target qt5).

Hope this helps.

tfoote, 2017-06-12:
With Gentoo being a rolling distribution, and keeping dependencies generally newer. It's likely that we won't be able to keep packaging for the full support cycles especially LTS releases if older dependencies are yanked out of the repositores. @allenh1 This is something that might be good to put together in a document about gentoo packaging.

allenh1, 2017-06-12:
@tfoote will do.

leio, 2017-06-12:
I think qt4 overall is fine for the time being; qtwebkit4 is a security problem, however.

Stuff isn't "yanked" if there are in-tree consumers (e.g if the relevant ROS stuff needing it specifically is in our main tree as packages too), but they are if they are security vulnerable and can be removed.
Because there are so many outdated consumers, sometimes things delay for too long while it would be nice if they didn't. But we eventually get to it and have to remove the consumers then too that aren't updated to security safe versions in a timely manner (where timely is rather relative and subjective to the consumer at hand).

Latest examples include:
* gstreamer:0.10
* webkit-gtk-2.4 (the old API before multi-process webkit2gtk; and thus also the last versions using gtk2) - in gentoo speak net-libs/webkit-gtk:2 and net-libs/webkit-gtk:3 (only webkit-gtk:4 is fine right now).
* qtwebkit:4

I did the comment on the PR because pocketsphinx and gstreamer0.10 is on my kill list and they will go away as soon as I have time to deal with it (after things like finally getting gstreamer 1.12 and gnome 3.24 in first..); https://github.com/gentoo/gentoo/pull/3321 - new ones into there is not OK to add to the main tree and I'm just warning here that they'll go away from main tree as well due to security issues. I don't know yet if pocketsphinx will be able to be upgraded to gstreamer1 instead or not, haven't gotten around to looking; but it for sure won't be working together with gstreamer0.10 within a month or two in Gentoo. gstreamer0.10 has not been maintained upstream for 3 years or so and varios CVE assigned bugs apply to it unfixed.

webkit-gtk-2.4 has only gotten some courtesy releases with easier to backport security fixes, but nothing anymore. Together with the stuff that wasn't backport fixed and all the new security vulnerabilities combined I believe comes to 300+ unfixed CVE assigned security bugs affecting webkit-gtk-2.4 versions. However you seem to use qtwebkit, which is in a similar position with the qt4 version and I believe also CVE count. PS: The gentoo mapping should define a slot, currently I see dev-qt/qtwebkit - it should be dev-qt/qtwebkit:4 or dev-qt/qtwebkit:5, former of which is in the Gentoo Qt teams kill list and consumers are actively being culled, until qtwebkit:4 itself can be removed as well. This became more achievable recently with announcement of old skype client getting retired (proprietary code user of qtwebkit:4 that was important enough to keep in some form and code couldn't be updated to qt5)

Alessandro-Barbieri, 2017-06-12:
@leio searched in Ubuntu and  libqtwebkit-dev is version 4, I'll explicitly add the slot soon

tfoote, 2017-08-11:
I think our default will be to keep rules available as long as they accurately represent what's available from the upstream distros. If there are any specific proposals to remove a package for known security issues they could be raised here. But I think as a policy we will want to push the security issues back upstream generally to get them fixed for the larger community. Removing the rosdep rule is only one of many ways to get the insecure package from upstream onto your ROS system if it's still available from the repositories.

closed 2017-08-11 by tfoote