ros/rosdistro #15497: [zesty only] libpcl-dev doesnt export the right VTK libraries

mikaelarguedas, 2017-07-14:
The `libpcl-dev` package doesnt export the right VTK libraries resulting on not being able to link against it.

Request for a rebuild has been [submitted upstream](https://bugs.launchpad.net/ubuntu/+source/pcl/+bug/1704459)

@wkentaro Once this is resolved you should be able to revert https://github.com/ros-perception/perception_pcl/commit/42683bd4069c3a8bcec56d6c4e686d9e8643e77b

@DLu this is the reason of the [recent failures](http://build.ros.org/job/Lbin_uZ64__costmap_2d__ubuntu_zesty_amd64__binary/1/console) of costmap_2d on zesty

mikaelarguedas, 2017-08-05:
@j-rivero @jspricke Does any of you know one of the listed ubuntu maintainers ? or what to change in the reported bug to get libpcl-dev rebuilt ? Thanks!

jspricke, 2017-08-08:
No idea, sorry. Try asking in IRC.

j-rivero, 2017-08-09:
> @j-rivero @jspricke Does any of you know one of the listed ubuntu maintainers ? 

AFAIK, there are no [Ubuntu maintainers for pcl](https://packages.ubuntu.com/source/zesty/pcl). The page displays the Debian ones (mainly Jochen and Leo). 

> or what to change in the reported bug to get libpcl-dev rebuilt ? Thanks!

We probably need to send a [Stable Release Update bug](https://wiki.ubuntu.com/StableReleaseUpdates) to trigger the whole process.

mikaelarguedas, 2017-08-09:
thanks you both for your answers! I'll look into getting an SRU request out

mikaelarguedas, 2017-08-28:
libpcl-dev has been rebuilt and released :tada: 

@wkentaro https://github.com/ros-perception/perception_pcl/pull/163

@matlabbe https://github.com/introlab/rtabmap/pull/228

closed 2017-08-28 by mikaelarguedas