ros/rosdistro #15806: CI requires alphabetical order for Ubuntu distributions

jatowler, 2017-08-22:
The CI script, the contribution guidelines, and anything else that relies on the Ubuntu distributions being in alphabetical order should be fixed to permit the best ordering, which is by version number.

With three exceptions, Ubuntu distributions [have been named in alphabetical order](https://en.wikipedia.org/wiki/Ubuntu_version_history#Table_of_versions). A happy consequence of this has been that sorting alphabetically also (with those exceptions) sorts chronologically.

Two of those exceptions happened in 2004 and 2005 (Warty Warthog was 4.10 and Hoary Hedgehog was 5.04) and can safely be ignored for the purposes of ROS. However, Artful Aardvark will be 17.10, but a naïve sort will place it at the top of the list, before 10.04 Lucid Lynx. This trend will presumably continue as Ubuntu distributions continue to run through the alphabet.

Therefore, the existing requirement that rosdeps be listed in alphabetical order, which has worked so far, is now counter-intuitive. The natural sort order for Ubuntu releases is version number, and that no longer coincides with alphabetical sorting. In addition, until now, all new releases have been added at the *bottom* of the list. But if the alphabetic requirement remains, new releases will be added in the *middle* of the list of distributions, which makes the list harder to parse to discover if the most-recent distribution is present, instead of just checking the bottom.

The [contribution guidelines](https://github.com/ros/rosdistro/blob/master/CONTRIBUTING.md) suggest that alphabetical order makes for better merging:

> Keep everything in alphabetical order for better merging.

However, this rule is too specific. A *consistent* ordering simplifies merging; alphabetical simply used to be convenient. This is no longer true.

The CI script, the contribution guidelines, and anything else that relies on the Ubuntu distributions being in alphabetical order should be fixed to permit the best ordering, which is by version number. This may require a manual list of versions somewhere, but that list will change every six months at most, and many rosdeps need to be updated every release anyway, so this requirement is negligible additional effort.

tfoote, 2017-08-22:
Maintaining the list of versions for Ubuntu is not too hard. However, rosdep keys will work for any OS/version, so it suddenly becomes much more complicated as we have to update it every time there's a new Debian version announced, or SUSE, or OSX, or anything else anyone else is managing. And in case the version is not explicitly listed we need to have a fallback behavior, which will likely be alphabetical. And if we fall behind on the official list, the CI will then it will require reordering keys when we update the list with the official order, which will cause more churn. The fallback behavior is required if the user is using user defined OS versions since those won't be on the list, unless we want to add the canonical sorting to the file format so that users can extend it without changing the source code.

As much as the ordering makes sense to be in chronological order, I believe that it's actually easier to read in alphabetical order. Humans are very well trained to find and read alphabetical lists. If I give you a specific codename to lookup, an alphabetical list is completely natural to read.

And although Ubuntu generally follows chronological order, debian definitely does not nor does osx. Using the built in sort is simpler and easier to specify generically. We would need to have very custom yaml loaders that were context aware of which block of yaml we're loading and inject the sort logic for loading and unloading.

Right now artful is at the top of the list and separated for zesty. However when Ubuntu B comes out. It will sit right next to artful, and Ubuntu C will be immediately after that. And although we want to support things for the long term. When we get back to the 2nd Ubuntu L that will be 13 years of support and we will have dropped support for lucid by then so we won't have to worry about interleaving.

As you stated it is necessary to be consistent. Sorting via a manually curated list is more complicated than is necessary. I'm strongly a -1 for changing the sorting since it will increase overhead on maintaining this repo. In particular explaining enforcing alphabetical ordering is very easy, versus explaining a custom order that is found in a config file that is based on a continually evolving external source, with some sort of fallback behavior. It will add overhead to accepting contributions from the community.

dirk-thomas, 2017-09-07:
Based on @tfoote comment I will close this issue. Please feel free to continue commenting on it.

closed 2017-09-07 by dirk-thomas