ros/rosdistro #15987: Does SDSM&T still supports Fedora Packaging?

leemgs, 2017-09-29:
I have found the repository that is released by SDSM&T at issue #4918 as fowlling:

[1] http://github.com/smd-ros-rpm-release/
[2] http://csc.mcs.sdsmt.edu/jenkins/
[3] http://github.com/cottsay/buildfarm/
[4] http://csc.mcs.sdsmt.edu/ros/rpmbuild/hydro.html
[5] http://csc.mcs.sdsmt.edu/ros/rpmbuild/indigo.html

However, I could not find proper repository to download a ROS Kinetic version for Fedora.
Do they still create RPM packages for ROS based on Fedora distribution?

\\CC: @cottsay

leemgs, 2017-09-29:
It seems that the **python-resdep** (=ROS System Dependency Installer) package has been hosted at https://src.fedoraproject.org/rpms/. Where can I find all rpm packages for ROS Kinetic on Fedora distribution? Welcome to any hint. 

[1] https://apps.fedoraproject.org/packages/python-rosdep
[2] https://src.fedoraproject.org/rpms/python-rosdep/blob/f23/f/python-rosdep.spec

**rosdep** (https://github.com/ros-infrastructure/rosdep) is a command-line tool for installing system dependencies. For end-users, rosdep helps you install system dependencies for software that you are building from source. For developers, rosdep simplifies the problem of installing system dependencies on different platforms. Instead of having to figure out which Debian package on Ubuntu Oneiric contains Boost, you can just specify a dependency on 'boost'.

leemgs, 2017-09-29:
In case of OpenSUSE, We could find some packages that were tried by some SUSE developers. 
* https://build.opensuse.org/search - search rpm packages "ros" keyword

In case of OpenSUSE, We could find some packages that were tried by some SUSE developers. 
* https://build.opensuse.org/search - search rpm packages "ros" keyword

leemgs, 2017-09-29:
In case of Fedora25.  There is  how to install ROS Kinetic on Fedora 25(tested for Fedora 25)
 * Reference: https://trac.fawkesrobotics.org/wiki/FawkesOnFedora
Install the following dependencies (in addition to the Fawkes dependencies).

```
sudo dnf install python-rosdep python-rosinstall_generator python-wstool python-rosinstall @buildsys-build python-qt5-devel python-defusedxml
```
Make qmake visible for the system by executing the following as admin
```
echo "export PATH=/usr/lib64/qt5/bin/:$PATH" > /etc/profile.d/qt5-qmake.sh
source /etc/profile
```
First create a working directory. The chosen format allows for well separable parallel installations of multiple versions.
```
mkdir -p /opt/ros/catkin_ws_kinetic
cd /opt/ros/catkin_ws_kinetic
```
Then download and build the packages. This will take some time and requires decent bandwidth.
```
rosinstall_generator desktop --rosdistro kinetic --deps --wet-only --tar > kinetic-desktop-wet.rosinstall
wstool init -j8 src kinetic-desktop-wet.rosinstall

rosinstall_generator navigation --rosdistro kinetic --deps --wet-only --tar > kinetic-navigation.rosinstall
rosinstall_generator ar_track_alvar --rosdistro kinetic --deps --wet-only --tar > kinetic-ar_track_alvar.rosinstall
wstool merge -t src kinetic-navigation.rosinstall
wstool merge -t src kinetic-ar_track_alvar.rosinstall
wstool update -t src
```
Next, build the whole workspace (you may also build with RelWithDebInfo if you want to be able to step into ROS code for debugging). We add the "-DENABLE_PRECOMPILED_HEADERS=OFF" flag to fix a ​build issue with OpenCV 3.
```
./src/catkin/bin/catkin_make_isolated --install --install-space=/opt/ros/kinetic \
  -DCMAKE_BUILD_TYPE=Release -DENABLE_PRECOMPILED_HEADERS=OFF
```
Add the setup.bash to your .bashrc
```
echo 'source /opt/ros/kinetic/setup.bash' >> ~/.bashrc
source ~/.bashrc 
```

leemgs, 2017-09-29:
\CC:   @myungjoo, @jijoongmoon, @sangjung.woo,  PTAL.

cottsay, 2017-09-29:
@leemgs, SDSM&T is no longer packaging ROS for Fedora. There was not enough support to keep the system healthy, so it was shut down.

Adding support to the newer docker-based ROS buildfarm would be a logical next step to restoring Fedora package support.

If you're interested, I suggest joinging the [Fedora Robotics SIG](https://fedoraproject.org/wiki/SIGs/Robotics).

ROS infrastructure packages, however, are packaged for Fedora (things like catkin-tools, rosdep, rosdistro, bloom, etc).

leemgs, 2017-09-29:
@cottday Thank you for sharing the status of SDSM&T packaging and Fedora Robotics SIG.  

BTW, Does no one release the RPM packaging of ROS currently?

closed 2018-01-13 by tfoote