ros/rosdistro #16138: Melodic Kick-Off Checklist

clalancette, 2017-10-23:
This is a checklist of things to do for the Melodic kick-off (pre development and development):
- [ ] Choose distribution name: Melodic
- [ ] commission a logo
- [ ] Add to REP3 with proposed platforms and minimum requirements, request feedback from community
- [ ] List on ROS wiki with expected release date (http://wiki.ros.org/Distributions)
- [ ] Add Ubuntu B + Debian Stretch to bootstrap repository on repos.ros.org
- [ ] Add Ubuntu B + Debian Stretch to buildfarm_deployment_config (note that one is not the live one it's for reference only)
- [x] Add Ubuntu B + Debian Stretch to ros_buildfarm/common.py

---
- [ ] Add newly supported platforms to list of platforms for each Python package (in `stdeb.cfg`), in order of dependency (do this as soon as debs for new Ubuntu are available, in pre-alpha!):
  - [ ] rospkg
  - [ ] catkin_pkg
  - [ ] rosdistro
  - [ ] rosdep
  - [ ] vcstools
  - [ ] wstool
  - [ ] rosinstall
  - [ ] bloom
  - [ ] rosinstall_generator
  - [ ] catkin_tools
  - [ ] vcstool
- [ ] Use [`ros_release_python`](https://github.com/ros-infrastructure/ros_release_python) to make the currently released versions available in the new distros. Use the `--include` option

---
- [ ] create Docker images for the new platforms for all targets https://github.com/osrf/multiarch-docker-image-generation/pull/1 (this seems to be generic now; do we need to do anything Melodic-specific here?)

---
- [ ] Create distribution files in ros/rosdistro
- [ ] add [ros_buildfarm_config](https://github.com/ros-infrastructure/ros_buildfarm_config) melodic entries
  - [ ] create new mailing list (https://groups.google.com/forum/#!forum/ros-buildfarm-melodic) with exactly the same settings as the existing ones
- [ ] generate new management jobs (see https://github.com/ros-infrastructure/ros_buildfarm/blob/master/doc/deploy_configuration.rst)

---
- [ ] [prerelease website](https://github.com/ros-infrastructure/prerelease_website)
  - [ ] Add new column in for the new distro with theme and placeholder image
  - [ ] Add new LTS distros as preferred
  - [ ] Make Ubuntu B the preferred OS
  - [ ] Test prerelease process (except for [latest ubuntu distros](https://github.com/ros-infrastructure/ros_buildfarm/issues/394))

---
- [ ] Add Melodic button to the [wiki](https://github.com/ros-infrastructure/roswiki)
- [ ] Add standard distribution wiki pages (planning, migration, etc...)
  - [ ] http://wiki.ros.org/melodic
  - [ ] http://wiki.ros.org/melodic/Planning
  - [ ] http://wiki.ros.org/melodic/Migration
    - [ ] `ros_comm`
  - [ ] http://wiki.ros.org/melodic/Planning/Maintenance
    - [ ] link to the blocked releases status page rather than using a static page
    - [ ] update previous distributions Maintenance pages accordingly
  - [ ] http://wiki.ros.org/melodic/Installation
    - [ ] Pages created http://wiki.ros.org/melodic/Installation
    - [ ] Fill out [Ubuntu](http://wiki.ros.org/melodic/Installation/Ubuntu) and [Debian](http://wiki.ros.org/melodic/Installation/Debian) Installation pages

---
*tag people once buildfarm is ready*
- [ ] Release `ros-base` packages into Melodic
  - [ ] catkin (@dirk-thomas)
  - [ ] genmsg (@dirk-thomas)
  - [ ] gencpp (@dirk-thomas)
  - [ ] geneus (@k-okada)
  - [ ] genlisp (@dirk-thomas)
  - [ ] genpy (@dirk-thomas)
  - [ ] gennodejs (@chris-smith @IanTheEngineer)
  - [ ] roscpp_core (@dirk-thomas)
  - [ ] message_generation (@dirk-thomas)
  - [ ] message_runtime (@dirk-thomas)
  - [ ] rosbag_migration_rule (@dirk-thomas)
  - [ ] std_msgs (@dirk-thomas)
  - [ ] ros_comm_msgs (@dirk-thomas)
  - [ ] cmake_modules (@wjwwood)
  - [ ] rospack (@dirk-thomas)
  - [ ] ros (@dirk-thomas)
  - [ ] roslisp (@dirk-thomas)
  - [ ] ros_comm (@dirk-thomas)
  - [ ] rosconsole_bridge (@dirk-thomas)
  - [ ] common_msgs (tfoote)
  - [ ] ros_tutorials (@dirk-thomas)
    - [ ] needs to be re-released as soon as the Melodic turtle is available
  - [ ] actionlib (mikaelarguedas)
  - [ ] angles (@tfoote)
  - [ ] bond_core (mikaelarguedas)
  - [ ] dynamic_reconfigure (mikaelarguedas)
  - [ ] class_loader (@mikaelarguedas)
  - [ ] pluginlib (mikaelarguedas)
  - [ ] nodelet_core (mikaelarguedas)
  - [ ] common_tutorials (wjwwood & stonier)

---
- [ ] Create melodic category on Discourse https://discourse.ros.org/c/release/melodic
- [ ] Announce availability of the Melodic build farm
- [ ] Update mirror documentation (see https://github.com/ros-infrastructure/mirror/commits/master for examples) 
  - [ ] update ros mirrors
  - [ ] update osrfoundation mirrors (wait until there are builds for ignition and gazebo for all supported platforms)
- [ ] Announce logo design and kick off t-shirt campaign

dirk-thomas, 2018-01-30:
I have released all `core` package up to `ros`. That already unblocks several other packages.

The next one for me is `ros_comm` which will likely take quite a while before that will happen due to the significant amount of pending tickets.

clalancette, 2018-01-30:
Excellent, thanks @dirk-thomas

dirk-thomas, 2018-02-05:
I have released `ros_comm` and `rosconsole_bridge`. All my remaining packages are now blocked on pending other releases.

mikaelarguedas, 2018-03-15:
@clalancette Initial release of `ros_base` packages has been done. Follow-up releases will come with potentially ABI/API breaking changes

hrnr, 2018-06-14:
Hi all,

unfortunately Melodic was released with bug in generated doxygen documentation. The docs are missing top navigation bar, making them hard to navigate. See  http://docs.ros.org/melodic/api/tf2_ros/html/c++/

The needed fix is https://github.com/ros-infrastructure/rosdoc_lite/pull/82. The rosdoc_lite is not receiving much attention, so I'm posting here to let @clalancette know.

Could someone please merge the fix and regenerate all the doxygen docs on http://docs.ros.org?

Cheers,
Jiri

tfoote, 2018-10-17:
@clalancette This looks to be hanging and the melodic launch is in the past. Can we close this out?