ros/rosdistro #16139: Melodic Release Checklist

clalancette, 2017-10-23:
This ticket is used to collect things that need to be done for the upcoming Melodic release. (copied from the [Lunar Release ticket](https://github.com/ros/rosdistro/issues/14112))

Existing Task Sets:
- [ ] Melodic kickoff list https://github.com/ros/rosdistro/issues/16138
- [ ] Melodic documentation audit
- [ ] Melodic community documentation audit

---
Documentation:
- [ ] Update verbiage on these wiki pages, e.g. 'upcoming release' vs. 'current release'
  - [ ] http://wiki.ros.org/melodic/Installation
  - [ ] http://wiki.ros.org/melodic
- [ ] Turtlesim icon
- [ ] N-turtle name

---
ROS Wiki
- [ ] make distroN the default distro
- [ ] Version macro for distroN needs to be updated from “Expected in distroN” to “distroN”
- [ ] Manually redeploy wiki once finished by pulling repo and restarting Apache instance
- [ ] Update symlink for default API doc:
  - /home/rosbot/docs/api → /home/rosbot/docs/distroN/api
  - /home/rosbot/latest → /home/rosbot/docs/distroN
- [ ] Update http://wiki.ros.org/Distributions
  - reflect released status (update from pending)
- [ ] Update http://wiki.ros.org/ROS/Installation to point to distroN

---
ROS browse
- [ ] update default distro

---
ROS docker images
- [ ] add Dockerfiles to create ROS docker images
---
Communication
- [ ] Announcement email to ros-users pointing to migration guide 
  - Post to ros-users
  - Post to blog
  - Post to G+
  - Post to Twitter

---
www.ros.org
- [ ] Update www.ros.org bottom row widget to point to new release with logo
- [ ] Update www.ros.org/install to point to new release with logo

clalancette, 2018-05-25:
Release is out!  Closing this.

closed 2018-05-25 by clalancette

dhood, 2018-06-20:
FYI we missed a step of adding melodic to http://wiki.ros.org/Installation/Ubuntu and other high-level pages  e.g. http://wiki.ros.org/action/info/Installation/Ubuntu?action=diff&rev2=6&rev1=5
Should all be done now; mentioning here so it is not forgotten the next release.

clalancette, 2018-06-20:
@dhood Thanks for the heads up.

mikaelarguedas, 2018-06-20:
Note: It's likely because of the load of moinmoin but this page now shows me C-Turlte install instructions and takes more than 10 sec to display only the content for melodic

tfoote, 2018-06-20:
Yeah, we're still transferring all of the content and it's only suppressed by the javascript that might be loading slowly for you remotely.