ros/rosdistro #16175: Request for including libccd-dev for indigo/trusty

wxmerkt, 2017-10-26:
As pointed out in #16125, in order to build up to date versions of FCL we require libccd-dev which is not included in Trusty (but is in Xenial). This issue is to import the debian into the ROS apt repository.

Relates to #16125

wxmerkt, 2017-10-30:
@tfoote What are the next steps here to discuss the inclusion of libccd-dev? Is there any formal procedure/tutorial I could follow to make the request beyond this issue?
Thank you

tfoote, 2017-10-30:
Being able to build the fcl_catkin package seems like a reasonable justification if the PPA is just providing a backport of a new package.

@wxmerkt Could you specify precisely which platforms and packages from which PPA you'd like us to import. We need a concrete proposal that we can review.

wxmerkt, 2017-10-30:
- The PPA to import from is https://launchpad.net/~libccd-debs, owned by @j-rivero 
- The package to import is ``libccd-dev`` for trusty
- Outside of fcl_catkin, newer versions of OMPL and DART use this PPA to provide the libccd dependency (version 1.5.0 is packaged as ``ros-indigo-libccd``, we need 2.0)

tfoote, 2017-10-31:
Thanks. Are there any conflicts with `ros-indigo-libccd`? How will this change behavior of anyone already relying on or using that existing package on trusty?

wxmerkt, 2017-10-31:
- I do not expect conflicts with libccd unless someone specifies ``libccd-dev`` in the dependency instead of ``libccd`` - but I may be wrong.
- Existing packages relying on rosdepkey ``libccd``: fcl (0.3.4), dart (using the above mentioned ppa - so perhaps not ros-indigo-libccd I suppose?) [build_depend](https://github.com/search?l=XML&q=%3Cbuild_depend%3Elibccd%3C%2Fbuild_depend%3E&type=Code&utf8=%E2%9C%93), [depend](https://github.com/search?utf8=%E2%9C%93&q=%22%3Cdepend%3Elibccd%3C%2Fdepend%3E%22&type=Code)
- Packages relying on rosdepkey ``libccd-dev``: 0 released, 2 private - [build_depend](https://github.com/search?utf8=%E2%9C%93&q=%3Cbuild_depend%3Elibccd-dev%3C%2Fbuild_depend%3E&type=Code), [depend](https://github.com/search?l=XML&q=%3Cdepend%3Elibccd-dev%3C%2Fdepend%3E&type=Code&utf8=%E2%9C%93)

j-rivero, 2017-11-07:
Introducing system packages which has other non ABI compatible versions delivered as a bloom/ROS package requires extra care. Potential points of conflicts:

1. The loader ends up loading a version of the library which was not the one used when the package was built depending if the use has source `setup.sh` or not. For this case using dart, having installed both licdd packages (system and ROS):

```shell
(trusty)jrivero@nium ~ $ ldd /usr/lib/libdart.so.5.1 | grep ccd
	libccd.so.2 => /usr/lib/libccd.so.2 (0x00007fb237539000)
(trusty)jrivero@nium ~ $ . /opt/ros/indigo/setup.bash 
(trusty)jrivero@nium ~ $ ldd /usr/lib/libdart.so.5.1 | grep ccd
	libccd.so.2 => /opt/ros/indigo/lib/libccd.so.2 (0x00007f2ccf9b0000)
```

2. The build system detect the wrong build modules with the same problem described in 1. for the loader.
```shell
(trusty)jrivero@nium ~ $ pkg-config --libs-only-L ccd
 
(trusty)jrivero@nium ~ $ . /opt/ros/indigo/setup.bash 
(trusty)jrivero@nium ~ $ pkg-config --libs-only-L ccd
-L/opt/ros/indigo/lib  
```

wxmerkt, 2017-11-07:
Thank you very much for the insight @j-rivero - what do you recommend in this case/what would be possible?

j-rivero, 2017-11-07:
> Thank you very much for the insight @j-rivero - what do you recommend in this case/what would be possible?

I see a couple of possible solutions, depending on the rest of the context specially with respect to fcl and moveit.

1. Create a `ros-indigo-libccd2` package that ships the code and make it conflict with `ros-indigo-libccd` so only one can be installed in the system.

2. Make some changes to generate a special version of libccd2 system package, in the same sense that we made for generating an [`fcl0.5` package](https://discourse.ros.org/t/fcl-0-5-packages-for-moveit-in-kinetic/364?u=jrivero). The changes should try to avoid the effects that I described above in the loader and pkg-config when both packages are installed in the same system.

wxmerkt, 2017-11-07:
Thank you very much for the detailed response. For now I've decided to remove fcl_catkin from indigo (#16298). The packages build successfully on Kinetic and we will continue source builds for indigo for now.

closed 2017-11-07 by wxmerkt

j-rivero, 2017-11-07:
> Thank you very much for the detailed response. For now I've decided to remove fcl_catkin from indigo (#16298). The packages build successfully on Kinetic and we will continue source builds for indigo for now.

Happy to help, thanks Wolfgang