ros/rosdistro #16181: ghprb_testing doesn't work for 3rdparty submissions

tfoote, 2017-10-27:
For security reasons, Travis will not pass the security tokens to jobs triggered by non-committers.

Talking about this the travis security concerns are valid. Otherwise malicious code could modify the test to expose the token contents.

Talking with @ruffsl and @mikaelarguedas we think the cleanest solution would be to switch to be a Jenkins job where we can explicitly run this element of the test not from the submitted branch but from the master or upstream with the token, which is something that I don't think we can configure on Travis.

Follow up to: https://github.com/ros/rosdistro/pull/16176 https://github.com/ros/rosdistro/commit/4e7a87a9de0ad7b8a13195feb8f1c93210771c6d#commitcomment-25212617

dirk-thomas, 2017-10-27:
> Talking with @ruffsl and @mikaelarguedas we think the cleanest solution would be to switch to be a Jenkins job where we can explicitly run this element of the test not from the submitted branch but [...]

I don't see why the different location of running the test would make any difference. The same security consideration still applied: a patch can expose the environment variable which contains the "secure" token.

mikaelarguedas, 2017-10-27:
I think the idea to mitigate this was to:
> run this element of the test not from the submitted branch but from the master or upstream with the token

So basically we would have a copy of our trusted rosdistro master branch and run the tests from that branch but run the tests against the patch provided by the user. One thing not detailed here is how do we plan on testing PRs modifying the tests themselves.

tfoote, 2017-10-27:
> One thing not detailed here is how do we plan on testing PRs modifying the tests themselves.

On Jenkins we have admin permissions and can tweak the job to change the branch that the tests pull from if we need to test changes to the tests themselves. Or we can manually reproduce locally with the token.

dirk-thomas, 2020-09-01:
Closing due to age and inactivity.

closed 2020-09-01 by dirk-thomas