ros/rosdistro #16336: Contributing guideline - tests and their dependencies.

peci1, 2017-11-11:
`CONTRIBUTING.md` says:

> To run the tests run ``nosetests`` in the root of the repository.
They require the rosdistro library, available on Ubuntu with the ROS repositories as python-rosdistro or via pip as rosdistro.

1. The tests actually need more dependencies than `rosdistro`. That should be somehow reflected in this text excerpt. Probably linking to .travis.yml for an idea about what to install?
1. When (after installing dependencies) I run `nosetests` in the root of repo, nothing happens, no tests are executed (well, noestests tell me 1 test ran, but the runtime is ~0 sec).

mikaelarguedas, 2017-11-11:
Thanks @peci1 for the report. Indeed it looks like we added several tests that require specific modules without adding them to the CONTRIBUTING.mg... Linking to the travis file is a good idea indeed this way we're sure the instructions stay up to date :+1: .

> When (after installing dependencies) I run nosetests in the root of repo, nothing happens, no tests are executed (well, noestests tell me 1 test ran, but the runtime is ~0 sec).

Hmm that's surprising, I just tried this in a fresh container and got the full suite of tests running:
```
# apt upgrade && apt install git dirmngr lsb-release
# sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
# apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116
# apt update && apt install python-pip
# pip install PyYAML argparse catkin_pkg ros_buildfarm rosdistro nose coverage yamllint unidiff rosdep PyGithub
# git clone https://github.com/ros/rosdistro
# cd rosdistro/
# nosetests -s
[...] # plenty of console output
Ran 6 tests in 107.476s

OK
```
Do you see any difference with what you tried on your side ?

peci1, 2017-11-11:
# cd rosdistro/
    # ls
    CONTRIBUTING.md  groovy  index.yaml  jade     lunar      releases  ros.key  test
    doc              hydro   indigo      kinetic  README.md  rosdep    scripts
    # nosetests -s
    .
    ----------------------------------------------------------------------
    Ran 1 test in 0.067s
    
    OK

peci1, 2017-11-11:
nosetests run in Python 2.7

peci1, 2017-11-11:
nosetests version 1.3.7

mikaelarguedas, 2017-11-11:
Unfortunately that is not enough information to help us figure out what your system setup is, can you provide the OS you are running, a pip freeze output as well as the history of commands you run before trying to run the test?

peci1, 2017-11-11:
Ubuntu 14.04.5 LTS 64bit, Python 2.7.6.  Runs on Intel i7 Haswell, Dell Latitude E6450.

The steps are the same as you did, just I didn't start with a fresh container, but my development machine.

actionlib==1.11.9
adium-theme-ubuntu==0.3.4
angles==1.9.11
apipkg==1.4
appdirs==1.4.3
apsw==3.8.2.post1
apt-xapian-index==0.45
autokey==0.90.4
autopep8==1.2.4
backports-abc==0.4
backports.shutil-get-terminal-size==1.0.0
backports.ssl-match-hostname==3.5.0.1
backports.weakref==1.0.post1
base-local-planner==1.12.13
baxter-interface==1.2.0
BeautifulSoup==3.2.1
bleach==1.5.0
bondpy==1.7.19
boto==2.43.0
Box2D==2.3.2
bzr==2.7.0
bzr-builddeb==2.8.6
BzrTools==2.6.0
camera-calibration==1.12.20
camera-calibration-parsers==1.11.12
camera-info-manager-py==0.2.3
catkin==0.6.19
catkin-lint==1.4.10
catkin-pkg==0.3.9
catkin-pkg-modules==0.3.9
catkin-tools==0.4.4
ccsm==0.9.11.3
certifi==2016.9.26
cffi==1.8.3
Chameleon==2.25
chardet==2.3.0
CherryPy==3.2.2
click==6.7
colorama==0.3.7
command-not-found==0.3
compizconfig-python==0.9.11.3
configglue==1.1.3.post0
configobj==5.0.6
configparser==3.5.0
controller-manager==0.9.4
controller-manager-msgs==0.9.4
CouchDB==0.8
coverage==4.4.2
cryptography==1.5.3
cssselect==0.9.1
cssutils==0.9.10
cv-bridge==1.11.15
cycler==0.10.0
Cython==0.25.1
debtagshw==0.1
decorator==4.0.10
defer==1.0.6
defusedxml==0.4.1
diagnostic-analysis==1.9.2
diagnostic-common-diagnostics==1.9.2
diagnostic-updater==1.9.2
dirspec==13.10
distro-info==0.12
dnspython==1.15.0
docutils==0.14
dulwich==0.15.0
[... 296 lines cut ...]
unity-lens-photos==1.0
urdfdom-py==0.3.3
urlgrabber==3.9.1
urllib3==1.19
uTidylib==0.2
vboxapi==1.0
vcstools==0.1.39
vizdoom==1.1.1
VTK==5.8.0
wadllib==1.3.2
wcwidth==0.1.7
webencodings==0.5
WebOb==1.3.1
Werkzeug==0.12.2
widgetsnbextension==1.2.6
wstool==0.1.13
wstools==0.4.4
wxPython==2.8.12.1
wxPython-common==2.8.12.1
xacro==1.9.5
xdiagnose===3.6.3build2
yamllint==1.10.0
zc.buildout==2.5.3
zenmap==6.40
zope.interface==4.3.2

mikaelarguedas, 2017-11-11:
Ok that's surprising.
I tried in trusty and without anything installed it runs fine.
Installing a few of the ros packages you have on your system (python-catkin-pkg and python-ros-buildfarm for example) I got nose to fail (not being found). Removing them from pip, installing it from apt and running nose from /usr/bin/nose worked.

Didnt get to the point where it runs but doesnt find any test yet though.

There is definitely something fishy when you have the same python packages installed from apt and from pip. I don't have much more time to look into it right now so I'd advise you to run it in a container for now or to try installing and run the python3 version of these packages to be sure to avoid weird mix-n-match of packages.

Please keep us updated if you find out where the problem comes from

peci1, 2017-11-11:
[debug_nose.txt](https://github.com/ros/rosdistro/files/1464216/debug_nose.txt)

A log from running nosetests in the rosdistro dir.

This was achived by the following .noserc:

    [nosetests]
    verbosity=5
    with-doctest=1

It seems only the test_url_validity test is picked up.

peci1, 2017-11-12:
Found the culprit!

It was the pip package nose-unittest, which installs a plugin that switches the testing to unittest mode. What's weird, according to the documentation I understood that should only happen when you pass "--with-unittest", which I don't do.

I also don't know how did the nose-unittest package get to my system, I don't use it directly anywhere, but maybe uninstalling it broke something.

Could you please test that you can reproduce my problem by installing nose-unittest?

mikaelarguedas, 2017-11-14:
Thanks @peci1 for investigating.
I can reproduce the issue. If I follow the steps in https://github.com/ros/rosdistro/issues/16336#issuecomment-343633404 I can get the tests to run.
If I install `nose-unittest` and run nosetests again I get the same behavior as you described in https://github.com/ros/rosdistro/issues/16336#issuecomment-343634242. (same thing if I istall the packages via apt but nose-unittest via pip)

PR #20563 (2019-03-15): Note issue with nose-unittest
Fixes #16336
commit 2497eda7 (2019-03-15): Note issue with nose-unittest (#20563)

Fixes #16336
closed 2019-03-15 by tfoote