ros/rosdistro #16337: Users who submit PRs to this repo are getting flagged "Abuse detected" by travis

tfoote, 2017-11-11:
This manifests itself in that it looks like the CI tests from travis never finish. They actually never start. 

They can be seen here: https://travis-ci.org/ros/rosdistro/requests

We have contacted Travis CI support in the past and they have either manually cleared the flag, and also advised that the user log into travis-ci.org so that they can verify the user somehow. 

I'm creating this to be a place to collect references and hopefully provide people with a solution. If you have to do the above login please comment here. 

Instances already observed include: #16335 #16186 #16299 #15650

tfoote, 2017-11-11:
Several of these have been first time contributors. However #16335 is particularly odd in that @peci1  successfully submitted a PR a few hours earlier.

peci1, 2017-11-11:
I logged in to travis. What next?

tfoote, 2017-11-11:
@peci1 I closed and reopened your PR and it's running the tests now.

tfoote, 2017-11-14:
I heard back from Travis CI Support and they don't have a workaround for this in the short term. They are aware of the pain and are working on a longer term solution. So I'll leave this open for visibility. If you have this problem when submitting to rosdistro please try logging into travis and then close and reopen the PR to retrigger travis as well as comment here on whether it works or doesn't work.

tfoote, 2017-12-06:
Hit by @YoshihiroMIYAKOSHI in #16536 
[image]

tfoote, 2019-01-23:
We haven't had any reports of this in a year, so I'm going to close this.

closed 2019-01-23 by tfoote

emersonknapp, 2019-07-26:
I logged in to travis, closed the PR, opened the PR, and got the same result the second time

https://github.com/ros/rosdistro/pull/21834