ros/rosdistro #16585: Rosdep update unable to process source

nkhedekar, 2017-12-14:
On a fresh installation of ubuntu 16.04, i am facing an error in doing rosdep update.

~$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/osx-homebrew.yaml
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml]:
	while scanning a quoted scalar
  in "<string>", line 4598, column 14:
            uri: 'https://raw.githubusercontent.c ... 
                 ^
found unexpected end of stream
  in "<string>", line 4598, column 68:
     ... rcontent.com/peci1/vrep_ros_brid
                                         ^
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml
Hit https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/ruby.yaml
ERROR: unable to process source [https://raw.githubusercontent.com/ros/rosdistro/master/releases/fuerte.yaml]:
	Failed to download target platform data for gbpdistro:
	Invalid GBP-distro/targets format: missing key: 'type'
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml

ERROR: Rosdep experienced an error: ''
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.8

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 137, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 381, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 389, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 579, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 469, in update_sources_list
    for dist_name in sorted(get_index().distributions.keys()):
  File "/usr/lib/python2.7/dist-packages/rosdep2/rosdistrohelper.py", line 69, in get_index
    _RDCache.index = rosdistro.get_index(_RDCache.index_url)
  File "/usr/lib/python2.7/dist-packages/rosdistro/__init__.py", line 101, in get_index
    yaml_str = load_url(url)
  File "/usr/lib/python2.7/dist-packages/rosdistro/loader.py", line 48, in load_url
    fh = urlopen(url, timeout=timeout)
  File "/usr/lib/python2.7/urllib2.py", line 154, in urlopen
    return opener.open(url, data, timeout)
  File "/usr/lib/python2.7/urllib2.py", line 429, in open
    response = self._open(req, data)
  File "/usr/lib/python2.7/urllib2.py", line 447, in _open
    '_open', req)
  File "/usr/lib/python2.7/urllib2.py", line 407, in _call_chain
    result = func(*args)
  File "/usr/lib/python2.7/urllib2.py", line 1241, in https_open
    context=self._context)
  File "/usr/lib/python2.7/urllib2.py", line 1201, in do_open
    r = h.getresponse(buffering=True)
  File "/usr/lib/python2.7/httplib.py", line 1136, in getresponse
    response.begin()
  File "/usr/lib/python2.7/httplib.py", line 453, in begin
    version, status, reason = self._read_status()
  File "/usr/lib/python2.7/httplib.py", line 417, in _read_status
    raise BadStatusLine(line)
BadStatusLine: ''

 I tried cloning the directory and pointing to the downloaded files from 20-default.list. This solves the end of stream errors but gives the following error.

$ rosdep update
reading in sources list data from /etc/ros/rosdep/sources.list.d
Hit file:///home/alpha/rosdistro/rosdep/osx-homebrew.yaml
Hit file:///home/alpha/rosdistro/rosdep/base.yaml
Hit file:///home/alpha/rosdistro/rosdep/python.yaml
Hit file:///home/alpha/rosdistro/rosdep/ruby.yaml
Hit file:///home/alpha/rosdistro/releases/fuerte.yaml
Query rosdistro index https://raw.githubusercontent.com/ros/rosdistro/master/index.yaml
Add distro "groovy"

ERROR: Rosdep experienced an error: while scanning a simple key
  in "<string>", line 4840, column 3:
      urdfd
      ^
could not found expected ':'
  in "<string>", line 4840, column 8:
      urdfd
           ^
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.8

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 137, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 381, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 389, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 579, in command_update
    error_handler=update_error_handler)
  File "/usr/lib/python2.7/dist-packages/rosdep2/sources_list.py", line 472, in update_sources_list
    rosdep_data = get_gbprepo_as_rosdep_data(dist_name)
  File "/usr/lib/python2.7/dist-packages/rosdep2/gbpdistro_support.py", line 146, in get_gbprepo_as_rosdep_data
    distro_file = get_release_file(gbpdistro)
  File "/usr/lib/python2.7/dist-packages/rosdep2/rosdistrohelper.py", line 76, in get_release_file
    dist_file = rosdistro.get_distribution_file(get_index(), distro)
  File "/usr/lib/python2.7/dist-packages/rosdistro/__init__.py", line 116, in get_distribution_file
    data = _get_dist_file_data(index, dist_name, 'distribution')
  File "/usr/lib/python2.7/dist-packages/rosdistro/__init__.py", line 194, in _get_dist_file_data
    data.append(_load_yaml_data(u))
  File "/usr/lib/python2.7/dist-packages/rosdistro/__init__.py", line 187, in _load_yaml_data
    return yaml.load(yaml_str)
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
    return loader.get_single_data()
  File "/usr/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
    node = self.get_single_node()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/lib/python2.7/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/lib/python2.7/dist-packages/yaml/parser.py", line 428, in parse_block_mapping_key
    if self.check_token(KeyToken):
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 170, in fetch_more_tokens
    return self.fetch_stream_end()
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 374, in fetch_stream_end
    self.remove_possible_simple_key()
  File "/usr/lib/python2.7/dist-packages/yaml/scanner.py", line 316, in remove_possible_simple_key
    "could not found expected ':'", self.get_mark())
ScannerError: while scanning a simple key
  in "<string>", line 4840, column 3:
      urdfd
      ^
could not found expected ':'
  in "<string>", line 4840, column 8:
      urdfd
           ^

nkhedekar, 2018-01-02:
Problem solved by reinstalling Ubuntu with GPT partition scheme for UEFI. Possible Cause - Using a MBR Partition Scheme for a computer using UEFI

closed 2018-01-02 by nkhedekar

tfoote, 2018-01-18:
I suspect that it was something else related to the reinstall not the partition scheme that resolve this issue. The partition scheme is unlikely to cause yaml parsing errors in the middle of the file. The first errors suggests that you're having network connectivity issues and possibly had a truncated or corrupted download of the yaml files which then is causing parse errors.

nkhedekar, 2018-01-27:
You are right.  It seems to be a network issue. I got the same error on a different system. The problem was solved by using a mobile hotspot.

Bibekshrestha, 2018-01-30:
why is osx homebrew.yaml in system with ubuntu?

tfoote, 2018-01-30:
The full database is downloaded on all platforms and queried locally. This allows things like testing and operations on non-native targets. Most of the rules are for ubuntu the homebrew ones are kept separate for easier maintainership.