ros/rosdistro #16764: Retiring old OS versions from rosdep keys

CBJamo, 2018-01-12:
Is there a policy for retiring old OS versions from the rosdep keys? 

There are numerous examples of a package name change long enough ago that the OS version with the old package name is no longer in support. For example:

```
libpq-dev:
  fedora:
    '21': [derelict-PQ-devel, postgresql-devel]
    '22': [derelict-PQ-devel, postgresql-devel]
    '23': [derelict-PQ-devel, postgresql-devel]
    '24': [derelict-PQ-devel, postgresql-devel]
    beefy: [derelict-postgresql-devel, postgresql-devel]
    heisenbug: [derelict-postgresql-devel, postgresql-devel]
    schrödinger’s: [derelict-postgresql-devel, postgresql-devel]
    spherical: [derelict-postgresql-devel, postgresql-devel]
```

Heisenbug (F20) has been out of support since mid-2015. If versions older than 21 were removed, then this entry could be simplified to just one key for Fedora. Scrolling through base.yaml I counted 67 packages (mostly in Fedora and Ubuntu) that could be simplified to a single key if out-of-support OS versions were removed. 

If there is no current policy, I would recommend removing keys for OS versions older than those listed in [REP 3](http://www.ros.org/reps/rep-0003.html).

tfoote, 2018-01-12:
Certainly at a minimum that any supported distro should be kept. However in general we bias towards keeping support if there might be anyone who's still using one of the platforms. The cost of maintaining a few extra lines in this file is relatively low compared to the cost of reproducing the older rules on a platform for which those lines were deleted.

There certainly is some point at which aging things out makes sense to clean up. And we more frequently will collapse rules where a package was added in distro X and for a while it was represented per version starting at that version so the older versions would not resolve it incorrectly. And we'll sometimes collapse those rules for simplicity, since that will not break anyone who was previously working, it will just mean if you try to resolve on an older unsupported system there will be new rules for packages that don't exist.

We have prototyped [new versions of the rules file](https://xylem.readthedocs.io/en/latest/spec.html) with better versioning support but the effort and disruption of deploying it has meant that we haven't pushed to deploy it.

CBJamo, 2018-01-17:
I like the direction Xylem is going, and am willing to put some work into it myself. If @wjwwood and/or @NikolausDemmel could provide some direction on what needs to be done, that would be appreciated.

[rosdep pr#573](https://github.com/ros-infrastructure/rosdep/pull/573) solves the core of this issue, so I'll close this. I'll open a pr here with cleaned up fedora keys once the rosdep changes are finalized.

closed 2018-01-17 by CBJamo