ros/rosdistro #16768: ros installation on debian:buster

weichm, 2018-01-13:
The installation on `debian:buster` didn't work as given in http://wiki.ros.org/lunar/Installation/Source, as some packages changed from `debian:stretch` to `debian:buster`.

So I had to change some commands from http://wiki.ros.org/lunar/Installation/Source: 

1) `apt-get install python-rosdep2 python-rosinstall python-rosinstall-generator python-wstool`, as `python-rosdep` changed to `python-rosdep2`.

2) `rosdep install --os=debian:stretch --from-paths src --ignore-src --rosdistro lunar -y`: explicitely give os as `stretch`, as there are no rules in https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml for `debian:buster`, but the `stretch` rules work for the problematic `libjpeg` and `libpcl-all`

3) I had to modify https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/python.yaml: the rule `python-rosdep --> python-rosdep` for `debian:buster` has to be `python-rosdep --> python-rosdep2`

If I can be of any help, I could create a pull-request for 2) and 3), but I don't know how to change documentation for 1)

BTW, the instructions on http://wiki.ros.org/lunar/Installation/Debian don't work neither for `buster`.

tfoote, 2018-01-14:
@weichm Lunar does not [target buster](http://www.ros.org/reps/rep-0003.html#lunar-loggerhead-may-2017-may-2019) so we likely will not update these instructions. 

@mikaelarguedas @clalancette FYI apparently buster has renamed the package to [python-rosdep2](https://packages.debian.org/buster/python-rosdep2) we'll likely see this coming up in people's source installations.

@j-rivero Is buster locked? Can we ask for this to be rolled back? It's not really accurate to call it rosdep2. That is the name of the module, but not of the tool.

j-rivero, 2018-01-17:
> @j-rivero Is buster locked? Can we ask for this to be rolled back? It's not really accurate to call it rosdep2. That is the name of the module, but not of the tool.

buster is the current `testing` distribution, a more stable stage than `unstable` and AFAIK is still far from being locked. Packages are migrated from `unstable` to `testing` automatically if they [fulfill some requirements](https://wiki.debian.org/DebianTesting). 

[Looking at the package](https://sources.debian.org/src/ros-rosdep/0.11.8-1/debian/changelog/), it seems to be maintained by @jspricke. Jochen: the only message related to the package name that I see in the [changelog](https://sources.debian.org/src/ros-rosdep/0.11.8-1/debian/changelog/) is related to problems with autopkgtest. What do you think about back to use the original name as Tully suggested? Let me know if you need  any help.

mikaelarguedas, 2018-01-21:
> FYI apparently buster has renamed the package to python-rosdep2 we'll likely see this coming up in people's source installations.

Ack

> If I can be of any help, I could create a pull-request for 2) and 3), but I don't know how to change documentation for 1)

@weichm we'll see if we can keep the rosdep package name consistent between stretch and buster, so no need to act on 1 and 3 for now.

For the packages that were missing in 2) when using `--os=debian:buster`, a PR(s) to the [rosdep package database](https://github.com/ros/rosdistro/tree/master/rosdep) would be much appreciated, guildelines for contributing can be found [here](https://github.com/ros/rosdistro/blob/master/CONTRIBUTING.md#rosdep-rules-contributions).

Thanks for reporting

jspricke, 2018-01-21:
I renamed it to match the module name (as the [Python policy ](https://www.debian.org/doc/packaging-manuals/python-policy/module_packages.html#package_names) proposes) and to fix the autopkgtest. But I'm not sure. If it' only meant as a command line tool, I would expect it in rosdep (and the implementation not exposed as a module). Is there a use for the module? @tfoote do you know the intention for the 2 in module name?

mikaelarguedas, 2018-01-22:
@tfoote @clalancette FYI
As a side note: the package has already been renamed in [Ubuntu artful and bionic](https://packages.ubuntu.com/search?keywords=python-rosdep2&searchon=names&exact=1&suite=all&section=all) so we will have a harder time renaming it in these distributions

tfoote, 2018-01-22:
The current implementation was a complete rewrite from the first version back in the early development. The rosdep2 module allowed Ken to develop it side by side. That first version was never really used externally and rosdep is really only designed to be used as a command line tool not as a library. There are a few places that we use it as a module but it's not a documented API. But if it's already in artful with the new name. The change has shipped and we'll have to deal with it.

tfoote, 2018-01-23:
I talked about this with @mikaelarguedas and since it's already been out in artful and we didn't notice it's likely that it will continue to be low impact. In particular all of our instructions start by suggesting adding the ROS apt sources before stating to install the ros packages. And as such as long as we package using the original name that package will resolve correctly for anyone in the ROS ecosystem.

The reason that this was run into on buster was that we had not yet released our python packages for buster, but they did exist upstream. So with tab completion the debian native version was found. Once we make releases into buster which @mikaelarguedas just opened PRs to support such as: https://github.com/ros-infrastructure/rosdep/pull/578 and the declared conflict https://github.com/ros-infrastructure/rosdep/pull/579 we should be back to a state that most people won't notice the different names once we release buster versions of the python tools, in particular rosdep.

mikaelarguedas, 2018-01-30:
@weichm ROS python packages (including rosdep) have been released for buster.
We didnt cut a new release of these packages so the [conflict declaration](https://github.com/ros-infrastructure/rosdep/pull/579) is not active yet, so please make sure to remove `python-rosdep2` from your system before testing.

Would you mind trying again the installation instructions from source ?

I was able to compile desktop-full in a fresh buster container following these instructions.

Note: I needed these keys to be able to install the dependencies for desktop full: https://github.com/ros/rosdistro/pull/16858

weichm, 2018-01-30:
@mikaelarguedas Sorry, as you said there is no new release... So I will try to compile using the `pip` package:
```
$ pip3 search rosdep
rosdep (0.11.8)  - rosdep package manager abstrction tool for ROS
```

mikaelarguedas, 2018-01-30:
@weichm you should be able to install it from the ros apt repository (no need to install from pip):
There is a note in the installation instructions to setup the sources: http://wiki.ros.org/lunar/Installation/Ubuntu#Installation.2BAC8-Ubuntu.2BAC8-Sources.Setup_your_sources.list
```
apt-get remove --purge python-rosdep2
sh -c 'echo "deb http://packages.ros.org/ros/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-latest.list'
apt-key adv --keyserver hkp://ha.pool.sks-keyservers.net:80 --recv-key 421C365BD9FF1F717815A3895523BAEEB01FA116
apt update
apt-get install python-rosdep python-rosinstall-generator python-wstool python-rosinstall build-essential
```

Then you should be good to continue with the rest of the instructions: http://wiki.ros.org/lunar/Installation/Source#Initializing_rosdep

mikaelarguedas, 2018-01-30:
@weichm I'm going to close this as from my understanding this is resolved. Feel free to comment here if anything goes wrong when compiling and we can reopen this issue.

closed 2018-01-30 by mikaelarguedas

weichm, 2018-01-30:
@mikaelarguedas Okay, I followed your instructions, trying to compile desktop full. However, I had to change
```
$ rosdep install --from-paths src --ignore-src --rosdistro lunar -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
rosconsole: No definition of [log4cxx] for OS version []
catkin: No definition of [python-argparse] for OS version []
...
```
to
```
$ rosdep install --os=debian:buster --from-paths src --ignore-src --rosdistro lunar -y
#All required rosdeps installed successfully
```

Somehow, I had to explicitly give the os by adding `--os=debian:buster`. Perhaps the error results from `/usr/lib/python2.7/dist-packages/rospkg/os_detect.py` (buster is missing):

```
class Debian(LsbDetect):

    def __init__(self, get_version_fn=None):
        super(Debian, self).__init__('debian', get_version_fn)

    def get_codename(self):
        if self.is_os():
            v = self.get_version()
            if v.startswith('7.'):
                return 'wheezy'
            if v.startswith('8.'):
                return 'jessie'
            if v.startswith('9.'):
                return 'stretch'
            return ''

```

As somewhere along the way, the file `/etc/os-release` seems to play a role, here are the contents:
```
$ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux buster/sid"
NAME="Debian GNU/Linux"
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

Right now, `./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release` is running ...

mikaelarguedas, 2018-01-30:
Good to know, thanks for testing it. I'll comment on the linked ticket