ros/rosdistro #16851: Continuous Integration/Release only for some packages of Meta-package.

awinkler, 2018-01-29:
I have a few ros packages http://wiki.ros.org/ifopt, for which the core package is solver independent and can be built with pure Eigen dependencies, while some other packages depend on external libraries that have to be installed manually. I want to have them documented on the wiki in the standard meta-package way as

[image]

Also, since e.g. ifopt_ipopt depends on IPOPT, i only want to do continuous integration on ifopt_core. How can this be achieved for individual packages of meta-package in the ``distribution.yaml`` file? 

    ifopt:
        doc:
          type: git
          url: https://github.com/ethz-adrl/ifopt.git
          version: master
        release:
          packages:
          - ifopt
          - ifopt_core
          - ifopt_ipopt
          - ifopt_snopt
          tags:
            release: release/indigo/{package}/{version}
          url: https://github.com/ethz-adrl/ifopt-release.git
          version: 1.0.0-0
        source:
          type: git
          url: https://github.com/ethz-adrl/ifopt.git
          version: master
        status: developed

dirk-thomas, 2018-01-29:
CI as well as documentation through the ROS buildfarm is being defined on a per-repository level. There is currently no way to only run a job on a subset of packages in a repository. If that is necessary for your case I would suggest to split the repository into two / multiple repos.

For repositories which you don't want CI to be run you can still have a `source` entry to support e.g. `rosinstall_generator` by adding `test_commits: false` to the `source` section (see example: https://github.com/ros/rosdistro/blob/946f1cba527be3b9ce2058671fb5a83b56a4a5b0/lunar/distribution.yaml#L619).

Having a metapackage which will result in the grouping in the wiki to appear is independent of the mapping to repositories. Commonly metapackages are in the same repo but there is no need for that.

awinkler, 2018-01-29:
thanks for the quick reply! Unfortunately multiple repos is not too sensible in my case, so I decided to just hack the CMakeLists of the projects that have external dependencies to exit when they don't find it.

closed 2018-01-29 by awinkler