ros/rosdistro #17100: Problem installing libpcl-dev for ros kinetic.

skamdar, 2018-03-10:
I tried sudo apt-get install libpcl-dev, here is what I got:

Preparing to unpack .../libpcl-dev_1.7.2-14build1_amd64.deb ...
Unpacking libpcl-dev (1.7.2-14build1) ...
dpkg: error processing archive /var/cache/apt/archives/libpcl-dev_1.7.2-14build1_amd64.deb (--unpack):
 trying to overwrite '/usr/include/pcl-1.7/pcl/for_each_type.h', which is also in package libpcl-common-1.7-dev 1.7.2-1+trusty6
dpkg-deb: error: subprocess paste was killed by signal (Broken pipe)
Errors were encountered while processing:
 /var/cache/apt/archives/libpcl-dev_1.7.2-14build1_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)

Please help!

mikaelarguedas, 2018-03-11:
It looks like you already installed a custom version of pcl on your machine.
`libpcl-common-1.7-dev` is not a package distributed by Ubuntu. You can try removing `libpcl-common-1.7-dev` and it's dependencies and then to install `libpcl-dev` again.

I'm going to close this as this is not related to ROS.
Please consider asking your ROS questions on  http://answers.ros.org following our support guidelines: http://wiki.ros.org/Support.
For non-ROS related questions, [stackoverflow](https://stackoverflow.com) or [askubuntu](https://askubuntu.com/) may be more appropriate platforms.

closed 2018-03-11 by mikaelarguedas

basavaraj1995, 2019-09-09:
`sudo apt-get install ros-kinetic-perception`