ros/rosdistro #17306: msg or srv document is not in the correct location

robotpilot, 2018-03-30:
I want to put msg or srv document of my message in 'http://docs.ros.org/api/' like below.
http://docs.ros.org/api/actionlib_msgs/html/index-msg.html

However, the document I put is not in the above address, but at 'http://docs.ros.org/kinetic/api/' like below
http://docs.ros.org/kinetic/api/turtlebot3_msgs/html/index-msg.html

If the document is not in the correct location, 404 error occurs on the ROS Wiki.

Do you have an option of Bloom setting that I missed?

dirk-thomas, 2018-03-30:
http://docs.ros.org/api/ points to the API docs for the latest released ROS distribution. Currently that is "Lunar". If you want the documentation to be available under that URL you have to make sure to register your package for the latest ROS distro. E.g. in May this will become "Melodic".

robotpilot, 2018-03-30:
@dirk-thomas 
It was a very curious part, and I understood your explanation.
If so, I will be ready for the next ROS Melodic version release.
Thank you very much. :)

closed 2018-03-30 by robotpilot