ros/rosdistro #17903: Drop Python 2.6 "support"

dirk-thomas, 2018-05-18:
`setuptools` has officially dropped support for Python 2.6 in Nov 2017 with [version 37](https://setuptools.readthedocs.io/en/latest/history.html#v37-0-0). Now we are starting to see the effect that "stuff" stops working with Python 2.6 (e.g. ros-infrastructure/rospkg#137).

Therefore I propose that we do remove Python 2.6 from all our Travis CI. Additionally we need to remove suites from the `stdeb.cfg` files to not release future versions into distros which don't have at least Python 2.7.

I don't know where the exact point of switch in Ubuntu was (the mailing list suggest around Natty) but I propose to remove Ubuntu distros before Precise.

@ros/ros_team If I could get a few comments / votes on this proposal I would go ahead an commit these changes across the various repositories containing Python packages and link the commit hashes here after the fact.

wjwwood, 2018-05-18:
+1, I've already removed 2.6 in favor of 2.7 in a few places, but also preserved it in others e.g. https://github.com/ros-infrastructure/rosdep/pull/593.

tfoote, 2018-05-18:
+1 for dropping distros older than precise from the stdebs. They are over 6 years old. We've had a goal of supporting older platforms so they can get updates etc. But this is a point when keeping the backwards compatibility will cause us a lot of effort we should prune it.

mikaelarguedas, 2018-05-18:
fine by me: dropping 2.6 and not releasing into anything older than precise :+1:

clalancette, 2018-05-18:
+1 seems reasonable to me

dirk-thomas, 2018-05-21:
After looking at many of our Python packages I decided to pick Oneiric as the lower boundary distro. Simply because that is what many packages which already target Python 2.7+ list.

dirk-thomas, 2018-05-21:
For Python 3.x I will also remove testing for Python 3.3 and older. 3.4 is being used by Trusty, 3.6 by Xenial, and 3.6 by Bionic.

dirk-thomas, 2018-05-21:
* ros-infrastructure/catkin_pkg#215
* ros-infrastructure/rosdep#609
* ros-infrastructure/rosdistro#120
* ros-infrastructure/rosinstall_generator#48
* ros-infrastructure/rospkg#140

All other Python packages already don't test for Python <2.7 anymore.

closed 2018-05-21 by dirk-thomas