ros/rosdistro #18152: rosdep crashes parsing entry for libglfw3-dev

MarcSeebold, 2018-06-07:
Hi,

Lately "rosdep update && rosdep db" gives us the following error:

`ERROR: Rosdep experienced an error: rosdep OS definition for [libglfw3-dev:ubuntu] must be a dictionary, string, or list: None
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.5

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 351, in _rosdep_main
    return _no_args_handler(command, parser, options, args)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 359, in _no_args_handler
    return command_handlers[command](options)
  File "/usr/lib/python2.7/dist-packages/rosdep2/main.py", line 701, in command_db
    inst_key, rule = d.get_rule_for_platform(os_name, os_version, installer_keys, default_key)
  File "/usr/lib/python2.7/dist-packages/rosdep2/lookup.py", line 140, in get_rule_for_platform
    raise InvalidData("rosdep OS definition for [%s:%s] must be a dictionary, string, or list: %s"%(self.rosdep_key, os_name, data), origin=self.origin)
InvalidData: rosdep OS definition for [libglfw3-dev:ubuntu] must be a dictionary, string, or list: None`

We think the problematic commit is https://github.com/ros/rosdistro/pull/18112 
It seems that `'*': [libglfw3-dev]` can't be parsed.

mikaelarguedas, 2018-06-07:
Thanks @MarcSeebold for reporting this.

Indeed the wildcart support has been added in [rosdep 0.12.0](https://github.com/ros-infrastructure/rosdep/blob/master/CHANGELOG.rst#0120-2018-02-07)

Can you try upgrading rosdep and see if the problem still happens ?

PS: you will need to install rosdep from the [ROS apt repository](http://wiki.ros.org/kinetic/Installation/Ubuntu#Installation.2BAC8-Ubuntu.2BAC8-Sources.Setup_your_sources.list) or via pip as the version in ubuntu is too old

MarcSeebold, 2018-06-07:
Thanks @mikaelarguedas for your very quick response!
I can verify that version "0.12.2" or rosdep works fine.
However, all our developers are stuck with version "0.11.5" that is available in Ubuntu14LTS apt repositories.
Adding a change like this to base.yaml breaks backward-compatibility.

Possible solutions I see:

1. Keep backward-compatibility of rosdistro with old rosdep versions
2. Fork rosdistro
3. Force users to upgrade to a more recent version of rosdep

dirk-thomas, 2018-06-07:
We ship the latest version of rosdep (0.12.2) to Precise: see http://repositories.ros.org/status_page/indigo_default.html?q=python-rosdep
So you should be able to pull the latest Debian package on Ubuntu 14 LTS without a problem.

MarcSeebold, 2018-06-07:
Thanks for the feedback. We'll upgrade to the latest rosdep version then.

closed 2018-06-07 by MarcSeebold