ros/rosdistro #18172: libpng12-dev does not exist on Debian Stretch

machinekoder, 2018-06-10:
See: https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L2519
The package does not exist anymore: https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=names&keywords=libpng12-dev

Probably replaced with `libpng-dev`

mikaelarguedas, 2018-06-10:
The `libpng-dev` key already maps to the right package on Debian Stretch and later: https://github.com/ros/rosdistro/blob/c891268cdd25c3ccdd5c5040d286586a23ece4a6/rosdep/base.yaml#L2495

The `libpng12-dev` key is now deprecated as this package disappeared from Ubuntu and Debian.

Please consider using the more generic `libpng-dev` in you package.xml

machinekoder, 2018-06-11:
In my ROS workspace it's OpenCV3 that depends on this package

https://github.com/ros-gbp/opencv3-release/blob/master/kinetic/package.xml#L19

Again, build does not fail since I have libpng-dev installed.

mikaelarguedas, 2018-06-11:
Does this mean that you are trying to build ROS Kinetic on Debian Stretch ?

This is not a supported combination.  Only ROS Lunar and Melodic target Debian Stretch.
Platforms targeted by each ROS distribution are listed on [REP3](http://www.ros.org/reps/rep-0003.html) 

On ROS Lunar, [opencv3 depends on `libpng-dev`](https://github.com/ros-gbp/opencv3-release/blob/f0a847a111e1d1835bc7aa985cf5067fcde2b21d/lunar/package.xml#L33) and not `libpng12-dev`. So I don;t think there is anything to change there.

Can you clarify what is the expected resolution of https://github.com/ros-gbp/opencv3-release/issues/18?

machinekoder, 2018-06-11:
I see, then let's close this issue.

closed 2018-06-11 by machinekoder