ros/rosdistro #18173: jasper does not exist anymore on Debian Stretch

machinekoder, 2018-06-10:
See: https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml#L1271

The package is not available in stretch:
https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=names&keywords=jasper

However, I did not encounter any error related this during compilation.

mikaelarguedas, 2018-06-10:
jasper has been removed from debian and ubuntu for security reasons:
https://bugs.launchpad.net/ubuntu/+source/jasper/+bug/1639737
https://web.archive.org/web/20160402170040/https://release.debian.org/transitions/html/jasper-rm.html

Not sure what it's been replaced with though and what library people commonly use for JPEG-2000.

It would be good to update the rules to be more dspecific so that users get notified that the rule doesnt have a resolution on the platforms where the package has been removed.

> However, I did not encounter any error related this during compilation.

It's possible if the package you compile doesnt actually need the jasper libraries.
Unfortunately that will be on a per package basis, we should look at all packages depending on that key to notify them the dependency cannot be resolved on recent Debian/Ubuntu platforms

machinekoder, 2018-06-11:
Looks like it's opencv3 that depends on `jasper` https://github.com/ros-gbp/opencv3-release/blob/master/kinetic/package.xml#L17

`libopenjp2` seems to replace on Debian: https://packages.debian.org/search?keywords=libopenjp2-7

machinekoder, 2018-06-11:
Ah okay, we are using Kinetic on Debian Stretch.

PR #18843 (2018-08-17): jasper is not available on recent ubuntu/debian
`libjasper-dev` was removed from debian https://web.archive.org/web/20160402170040/https://release.debian.org/transitions/html/jasper-rm.html
It is not available on Debian Stretch and later, also not available on Ubuntu Zesty and later.

Closes #18173
closed 2018-08-17 by mikaelarguedas