ros/rosdistro #18272: git-core in bionic

BillWSY, 2018-06-26:
I am getting this error:
```
00:06:42 Found the following packages:
00:06:42   - grpc
00:06:42 Package maintainer emails: shengye@ucsd.edu
00:06:44 Resolved the dependencies to the following binary packages:
00:06:44   - ros-melodic-catkin
00:06:44 Always install the following generic dependencies:
00:06:44   - build-essential
00:06:44   - python3
00:06:44   - ros-melodic-catkin
00:06:44 Identified the following build dependencies (ignoring packages available from source):
00:06:44   - autoconf
00:06:44   - catkin
00:06:44   - git
00:06:44   - libtool
00:06:44   - rsync
00:06:44   - zlib
00:06:44 Resolved the dependencies to the following binary packages:
00:06:44   - autoconf
00:06:44   - git-core
00:06:44   - libltdl-dev
00:06:44   - libtool
00:06:44   - libtool-bin
00:06:44   - ros-melodic-catkin
00:06:44   - rsync
00:06:44   - zlib1g-dev
00:06:44 Traceback (most recent call last):
00:06:44   File "/usr/lib/python3/dist-packages/apt/cache.py", line 257, in __getitem__
00:06:44     return self._weakref[key]
00:06:44   File "/usr/lib/python3.6/weakref.py", line 137, in __getitem__
00:06:44     o = self.data[key]()
00:06:44 KeyError: 'git-core'
```
http://build.ros.org/job/Mdev__grpc__ubuntu_bionic_amd64/1/display/redirect

Is that because git-core is removed from bionic? https://launchpad.net/ubuntu/bionic/amd64/git-core

tfoote, 2018-06-27:
`git-core` has been moved to be a virtual package in bionic: https://packages.ubuntu.com/bionic/git-core this is an issue I've tickted here: https://github.com/ros-infrastructure/ros_buildfarm/issues/557

closed 2018-06-27 by tfoote

tfoote, 2018-06-28:
@BillWSY Follow up in #18305