ros/rosdistro #18362: Errors in authorization.

hakuturu583, 2018-07-04:
I posted pull request through bloom.(https://github.com/ros/rosdistro/pull/18357)
My code was passed travis test but I seems it was failed when 
```
verifying diff of file '/home/travis/build/ros/rosdistro/kinetic/distribution.yaml'
```
Authorization publickey was denied.
```
Warning: Permanently added the RSA host key for IP address '192.30.253.113' to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.
Please make sure you have the correct access rights
and the repository exists.
Warning: Permanently added the RSA host key for IP address '192.30.253.112' to the list of known hosts.
Permission denied (publickey).
fatal: Could not read from remote repository.
Please make sure you have the correct access rights
and the repository exists.
ERROR: In file '''/home/travis/build/ros/rosdistro/kinetic/distribution.yaml''': Could not validate source entry for repo cog_publisher with error [[[Could not validate repository with url git@github.com:OUXT-Polaris/cog_publisher-release.git and version master from entry at line 1487]]]
ERROR: In file '''/home/travis/build/ros/rosdistro/kinetic/distribution.yaml''': Could not validate doc entry for repo cog_publisher with error [[[Could not validate repository with url git@github.com:OUXT-Polaris/cog_publisher-release.git and version master from entry at line 1478]]]
F
======================================================================
FAIL: test.rosdistro_check_urls_test.test_rosdistro_urls
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/travis/virtualenv/python2.7.14/lib/python2.7/site-packages/nose/case.py", line 197, in runTest
    self.test(*self.arg)
  File "/home/travis/build/ros/rosdistro/test/rosdistro_check_urls_test.py", line 29, in test_rosdistro_urls
    assert not failed_distros, "There were problems with urls in the distribution files for these distros: %s, see folded blocks above for details: %s" % (failed_distros, fold_blocks)
AssertionError: There were problems with urls in the distribution files for these distros: ['kinetic'], see folded blocks above for details: ['block4']
-------------------- >> begin captured logging << --------------------
rosdistro: DEBUG: Load index from "file:///home/travis/build/ros/rosdistro/index.yaml"
rosdistro: DEBUG: Load index from "file:///home/travis/build/ros/rosdistro/index.yaml"
rosdistro: DEBUG: Load file from "file:///home/travis/build/ros/rosdistro/indigo/distribution.yaml"
rosdistro: DEBUG: Load index from "file:///home/travis/build/ros/rosdistro/index.yaml"
rosdistro: DEBUG: Load file from "file:///home/travis/build/ros/rosdistro/kinetic/distribution.yaml"
rosdistro: DEBUG: Load index from "file:///home/travis/build/ros/rosdistro/index.yaml"
rosdistro: DEBUG: Load file from "file:///home/travis/build/ros/rosdistro/lunar/distribution.yaml"
rosdistro: DEBUG: Load index from "file:///home/travis/build/ros/rosdistro/index.yaml"
rosdistro: DEBUG: Load file from "file:///home/travis/build/ros/rosdistro/melodic/distribution.yaml"
--------------------- >> end captured logging << ---------------------
```

Please tell me how to solve this problem.
My detailed travis test result is [here](https://travis-ci.org/ros/rosdistro/builds/400028525#L6812).

mikaelarguedas, 2018-07-04:
Can you try updating your pull request to use the https url of your repository instead of the ssh one?

dirk-thomas, 2018-07-05:
The error message in the Travis log states:

```
Skipped "github_manifest_provider()": Could not git ls-remote repository "git@github.com:OUXT-Polaris/cog_publisher-release.git"
Skip non-bitbucket url "git@github.com:OUXT-Polaris/cog_publisher-release.git"
Skipped "bitbucket_manifest_provider()": Cannot handle non bitbucket url.
Skipped "git_manifest_provider()": Unable to fetch package.xml: Could not clone repository "git@github.com:OUXT-Polaris/cog_publisher-release.git" at reference "release/kinetic/cog_publisher/1.0.1-2"
```

As @mikaelarguedas suggested already you must use `https` urls instead of `git@github.com` (see all other URLs in the distribution file).

tfoote, 2018-07-06:
Yes, all repos need to be accessible via https not ssh for the buildfarm to be able to operate on them. You changed one of 3 urls to be https. All 3 need to be https.

closed 2018-07-06 by tfoote