ros/rosdistro #18406: Distro versions with '*' not working?

Rayman, 2018-07-10:
Hey, I tried to depend on `python-seaborn` in my package, but I got the following error. Are distro versions with a '*' supported? Because it doesn't work for me:
```sh
$ rosdep resolve python-seaborn
```
```
No definition of [python-seaborn] for OS version [xenial]
	rosdep key : python-seaborn
	OS name    : ubuntu
	OS version : xenial
	Data: arch:
- python2-seaborn
debian:
- python-seaborn
fedora:
  pip:
    packages:
    - seaborn
gentoo:
- dev-python/seaborn
ubuntu:
  '*':
  - python-seaborn
  trusty:
    pip:
      packages:
      - seaborn

```
```
rosdep --version
0.11.8
```

mikaelarguedas, 2018-07-10:
This look like a duplicate of https://github.com/ros/rosdistro/issues/18152
Have you tried updating rosdep?
You need at least version 0.12.0 to parse these rules

tfoote, 2018-07-10:
Closing as duplicate of #18152 upgrading rosdep should resolve the issue.

closed 2018-07-10 by tfoote