ros/rosdistro #18956: Rosdep to python-requests

yotabits, 2018-08-27:
On ubuntu 16.04(Kinetic) I would like to install python-request package as a dependency of my package
but I got python3-requests installed instead. I think it's because of the content of python.yaml file under rosdistro folder
eg: 
```
python-requests:
  debian: [python-requests]
  fedora: [python-requests]
  gentoo: [dev-python/requests]
  osx:
    pip:
      packages: [requests]
  ubuntu:
    '*': [python-requests]
    trusty_python3: [python3-requests]
    utopic_python3: [python3-requests]
    vivid_python3: [python3-requests]
    wily_python3: [python3-requests]
    xenial_python3: [python3-requests]
    yakkety_python3: [python3-requests]
    zesty_python3: [python3-requests]
```
It seems like the wild card set for ubuntu does not work properly for me.

In fact when I run `rosdep resolve python-requests` I have the following on xenial:
`ERROR: No definition of [python-requests] for OS version [xenial]`
On Trusty I have the following:
```
#apt
python-requests
```

**BUT**

If I set rosdep python.yml to https://raw.githubusercontent.com/yotabits/rosdistro/master/rosdep/python.yaml (where python-requests is specifically defined for xenial and xenial_python3) `rosdep resolve python-requests` gives me the same output on xenial and trusty 
Eg:
```
#apt
python-requests
```
 
So here are my questions:
- How does the *_python3 entry works from the rosdep pov?
- How can I have python-requests package installed instead of python3-requests (Eg: have the wildcard properly used)?

My questions may be trivial but I cannot find an answer so far, any help would be appreciated.

dirk-thomas, 2018-08-28:
In a Xenial docker container `rosdep resolve python-requests` correctly resolves to `python-requests`.

I assume you are simply using an older version of `rosdep` which doesn'y have support for the wildcard entry. And therefore it fails to resolve the key for "xenial". Please make sure that you are using the latest version (currently 0.12.2).

If this resolves your problem please close this ticket.

yotabits, 2018-08-29:
mea culpa,

My apt-mirror of packages.ros.org was not up to date.

closed 2018-08-29 by yotabits