ros/rosdistro #19295: grpc conflicting versions

tfoote, 2018-10-17:
grpc is released into kinetic and melodic but that is going to conflict with the system version. 

This was flagged in #19184 and  talking with the maintainer @BillWSY we talked about reducing the ROS packages to just provide the extras that are not in the upstream repositories: https://github.com/ros/rosdistro/pull/19184#issuecomment-424827756

For reference the Melodic PR: #19185

And an incoming rosdep key which will trigger the collision #19292

@BillWSY Would you be able to switch this to the stripped down version soon?

tfoote, 2018-10-17:
The problem is that we cannot let users choose between the versions as it will just cause problems for downstream users. 

If package A chooses implementation 1 and package B choose implementation 2. If a 3rdparty builds package C and depends on package A and package B it will be getting symbols from both conflicting versions but only one can be linked in so you'll get at best linking failures at worst non-deterministic runtime failures. 

@nuclearsandwich Looking at it a little closer I think that the python implementation in #19292 doesn't appear to rely on the grpc library I think it has a pure implementation so that's actually slightly separated.

BillWSY, 2018-10-17:
That makes sense. I think ROS had similar issues about OpenCV in indigo, but of course, this is much easier to solve than OpenCV.

The problem for me is that our robot is still running indigo, which makes supporting kinectic/melodic and testing them a lower priority to me. I will check with my colleagues @ruffsl and @MaidouPP to see if they are willing to work on this. Otherwise, I will try to get it done myself in a month or two.

tfoote, 2020-01-02:
I'm going to close this in favor of tracking it at: https://github.com/CogRob/catkin_grpc/issues/35

closed 2020-01-02 by tfoote