ros/rosdistro #19312: librealsense won't install successfully on 16.04.5

tfoote, 2018-10-19:
This is due to the upgraded LTS kernel. 

There's a tracking issue here: https://github.com/IntelRealSense/librealsense/issues/2402

If we can't get this installable we'll likely need to strip this as a default installation element from the turtlebot stack and make it an optional addon.

gavanderhoorn, 2018-11-04:
> If we can't get this installable we'll likely need to strip this as a default installation element from the turtlebot stack and make it an optional addon.

please do.

I've had to help quite a few people here in the lab that broke their ROS installation by installing all of the turtlebot 2 stack on it and have it fail with `librealsense` and not recover.

tfoote, 2018-11-08:
@gavanderhoorn I tested it on a fresh 16.04.5 machine and it appeared to not block the system from installing other packages, just that package was errored: https://github.com/IntelRealSense/librealsense/issues/2402#issuecomment-432045561 Are you seeing different behavior?

musamarcusso, 2019-02-28:
I have seen this behaviour in Docker images with Ubuntu 16.04. The image doesn't build because of this require modifications on the package configuration and they fail, like already mentioned. I ended up making a custom with some fixes and ignoring the rosdep installation using `skip_keys` but it was still not working until last time I tried.

gtoff, 2019-08-29:
> I have seen this behaviour in Docker images with Ubuntu 16.04. The image doesn't build because of this require modifications on the package configuration and they fail, like already mentioned. I [...]

Same here. We're still having trouble with librealsense in all our containers since the post-install script fails to detect it's running in a container

LucidOne, 2019-10-17:
I spent some time trying to [fix it](https://github.com/IntelRealSense/librealsense/issues/2402#issuecomment-541843254) but at this point I think it makes more sense to pull the dependency until the problem is resolved.

dirk-thomas, 2020-09-01:
Upstream ticket marked as resolved.

closed 2020-09-01 by dirk-thomas