ros/rosdistro #19394: Inconsistent Qwt6 and Qt versions

pgorczak, 2018-11-01:
I'm working on adding a key for Qwt6 with Qt5 for [rqt_multiplot](http://wiki.ros.org/rqt_multiplot) (see #19329 for prior discussion). While checking out the state of Qt4/Qt5 across distros I found some inconsistencies. It seems like Arch and Gentoo have switched to Qt5 under the hood while keeping the same package name.

Before opening a PR adding a new libqwt6-qt5 key I'd like to check if there's a way to reflect the situation in Arch and Gentoo e.g. by adding a version range to the dependencies.

## Qwt6 and Qt versions across distros

* [Arch Linux][arch] uses Qt4 until 6.1.3-1 and then switches to Qt5.
* Debian has [libqwt-dev][debian-4] built with Qt4 and [libqwt-qt5-dev][debian-5] built with Qt5.
* Fedora also as [qt4][fedora-4] and [qt5][fedora-5] versions.
* Gentoo seems like it had support for both until 6.1.3-r1 but the reference to qt4 has recently been [dropped][gentoo] with 6.1.3-r2
* [Ubuntu][ubuntu] has libqwt-dev built with Qt4 and libqwt-qt5-dev built with Qt5.

[arch]: https://git.archlinux.org/svntogit/packages.git/log/trunk?h=packages/qwt
[debian-4]: https://packages.debian.org/stretch/libqwt-dev
[debian-5]: https://packages.debian.org/stretch/libqwt-qt5-dev
[fedora-4]: https://apps.fedoraproject.org/packages/qwt-devel
[fedora-5]: https://apps.fedoraproject.org/packages/qwt-qt5-devel
[gentoo]: https://gitweb.gentoo.org/repo/gentoo.git/commit/x11-libs/qwt?id=173b46182ec71ef2a555875c806d46a822f41c40
[ubuntu]: https://packages.ubuntu.com/search?keywords=libqwt

allenh1, 2018-11-27:
@pgorczak For Gentoo, we recently added the Qt4 dependencies back to the repo (since Gentoo pulled them). I'm not really familiar with the Qwt library, but we can certainly modify the version to which the Gentoo one points. My usual preference is to aim for the most recent version that is compatible with the rest of the environment.

samuelba, 2018-11-30:
I created a pull request https://github.com/ros/rosdistro/pull/19567 since rqt_multiplot builds started failing after commit https://github.com/ros/rosdistro/commit/25747529d24ca067c148149cb6c6fc34f676be84 .

pgorczak, 2018-11-30:
@allenh1 in https://github.com/ros/rosdistro/pull/19327 and https://github.com/ros/rosdistro/pull/19329 we chose to make the Qt4/Qt5 switch explicit by adding a new key as needed. When https://github.com/ros/rosdistro/pull/19567 gets merged, we should probably change the Gentoo *libqwt6* key to point at a version using Qt4.

samuelba, 2018-12-12:
@pgorczak I guess with can close this since pull request https://github.com/ros/rosdistro/pull/19567 got merged?