ros/rosdistro #19481: Premature mumber 'control.tar.xz' before 'control.tar.gz'

ziangliuusc, 2018-11-16:
I'm tring to install ROS on my Ubuntu 14.04LTS, but the rosdistro package seems unable to be installed. The error message is as following:  

dpkg-deb: error: archive '/var/cache/apt/archives/python-rosdistro-modules_0.7.0-1_all.deb' has premature member 'control.tar.xz' before 'control.tar.gz', giving up
dpkg: error processing archive /var/cache/apt/archives/python-rosdistro-modules_0.7.0-1_all.deb (--unpack):
 subprocess dpkg-deb --control returned error exit status 2
dpkg-deb: error: archive '/var/cache/apt/archives/python-rosdistro_0.7.0-100_all.deb' has premature member 'control.tar.xz' before 'control.tar.gz', giving up
dpkg: error processing archive /var/cache/apt/archives/python-rosdistro_0.7.0-100_all.deb (--unpack):
 subprocess dpkg-deb --control returned error exit status 2

Is it a issue with the packages or with my system? Newbie to Linux and ROS, it been painful trying to get everything to work. I have Linux Kernel 4.2.1 and Python 2.7 installed.

tfoote, 2018-11-16:
This is a regression due to the package being built with a newer version of the debian package format that trusty doesn't know about. 

@dirk-thomas @nuclearsandwich FYI it looks like we didn't use the older format for this latest release.

tfoote, 2018-11-16:
Actually sorry, the fix for this has been backported to trusty. You need to make sure that you have the latest version of dpkg. https://bugs.launchpad.net/ubuntu/+source/dpkg/+bug/1730627/comments/19

https://github.com/ros-infrastructure/ros_buildfarm/pull/564

closed 2018-11-16 by tfoote

ziangliuusc, 2018-11-16:
Okay thank you!

knorth55, 2018-11-29:
Currently, `trusty-security` stops at `1.17.5ubuntu5.6`.

```bash
$ apt-cache policy dpkg
dpkg:
  Installed: 1.17.5ubuntu5.8
  Candidate: 1.17.5ubuntu5.8
  Version table:
 *** 1.17.5ubuntu5.8 0
        500 http://archive.ubuntu.com/ubuntu/ trusty-updates/main amd64 Packages
        100 /var/lib/dpkg/status
     1.17.5ubuntu5.6 0
        500 http://security.ubuntu.com/ubuntu/ trusty-security/main amd64 Packages
     1.17.5ubuntu5 0
        500 http://archive.ubuntu.com/ubuntu/ trusty/main amd64 Packages
```
In order to install `dpkg` `1.17.5ubuntu5.8`, you need to enable `trusty-updates`
[image]