ros/rosdistro #19566: libqwt-dev misses QT5 for xenial and above

samuelba, 2018-11-30:
@pgorczak  Commit https://github.com/ros/rosdistro/commit/25747529d24ca067c148149cb6c6fc34f676be84 broke the rosdep key `libqwt-dev` for rqt_multiplot. Since releases for xenial and above will need QT5. Either these changes should be reverted or we have to add a new key like  `libqwt-qt5-dev` as suggested by https://github.com/ros/rosdistro/issues/19394 .

pgorczak, 2018-11-30:
This happened due to https://github.com/ros/rosdistro/pull/19327 and https://github.com/ros/rosdistro/pull/19329 where we decided to not switch to Qt5 under the hood for xenial and above. I think your PR https://github.com/ros/rosdistro/pull/19567 is the way to go.

samuelba, 2018-12-12:
Fixed with pull request https://github.com/ros/rosdistro/pull/19567

closed 2018-12-12 by samuelba