ros/rosdistro #19845: npm is uninstallable on stretch

tfoote, 2019-01-03:
This rosdep key is invalid for stretch: https://github.com/ros/rosdistro/blob/990521767b21c1917c03df828a9ab67843e453e3/rosdep/base.yaml#L4050

It appears that due to a bit of a packaging snafu it was broken and was pulled instead of being upgraded at final testing.

Some of the threads: 
- https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=857986
- https://lists.debian.org/debian-release/2017/04/msg00115.html

The net result was that it was pulled from Stretch. What I don't understand is why it couldn't be added back later...

It's available in stretch-backports https://packages.debian.org/stretch-backports/npm

@clalancette @nuclearsandwich How do you feel about us bringing in the backport to the ROS repo.

NickZ, 2019-01-03:
Would it also be possible to set the `nodejs` target to be the one in backports as well? The one in the main repo is 4.8, and the one in backports is 8.11.1. My package will not build on 4.8.

clalancette, 2019-01-17:
> @clalancette @nuclearsandwich How do you feel about us bringing in the backport to the ROS repo.

We discussed this a couple of days ago, but I think the risk here is that we end up bringing in other, random stuff from backports.

We also discussed pulling npm in as a package that we host on ROS infrastructure; that has the benefit of not pulling in unwanted stuff from backports, but has the downside that we'll have to manually pick up any security updates for npm.  Still, that might be the right way to go here.  Further thoughts here?

tfoote, 2019-01-18:
I think it makes sense to specifically cherry pick the npm backport into our repo so that we can just have coverage across all platforms. If the backport starts looking like we have to change versions of many other packages(like nodejs) then I think it won't make sense and we'll just have to blacklist anything npm based on stretch.

After some debugging it looks like the list is much longer than I had hoped.
```
The following packages have unmet dependencies:
 npm : Depends: nodejs (>= 6.11~) but 4.8.2~dfsg-1 is to be installed
       Depends: node-abbrev (>= 1.1.1~) but 1.0.9-1 is to be installed
       Depends: node-ansi-regex (>= 3.0~) but 2.0.0-1 is to be installed
       Depends: node-cacache (>= 10.0.4~) but it is not going to be installed
       Depends: node-config-chain (>= 1.1.11~) but it is not going to be installed
       Depends: node-glob (>= 7.1.2~) but 7.1.1-1 is to be installed
       Depends: node-hosted-git-info (>= 2.6~) but 2.1.5-1 is to be installed
       Depends: node-ini (>= 1.3.5~) but 1.1.0-1 is to be installed
       Depends: node-npm-package-arg but it is not going to be installed
       Depends: node-jsonstream (>= 1.3.2~) but 1.0.3-4 is to be installed
       Depends: node-libnpx (>= 10.0.1~) but it is not going to be installed
       Depends: node-lockfile (>= 1.0.3~) but 0.4.1-1 is to be installed
       Depends: node-lru-cache (>= 4.1.1~) but 4.0.2-1 is to be installed
       Depends: node-move-concurrently (>= 1.0.1~) but it is not going to be installed
       Depends: node-normalize-package-data (>= 2.4~) but 2.3.5-2 is to be installed
       Depends: node-gyp (>= 3.6.2~) but 3.4.0-1 is to be installed
       Depends: node-resolve-from (>= 4.0~) but 2.0.0-1 is to be installed
       Depends: node-npmlog (>= 4.1.2~) but 0.0.4-1 is to be installed
       Depends: node-osenv (>= 0.1.5~) but 0.1.0-1 is to be installed
       Depends: node-read-package-json (>= 2.0.13~) but 1.2.4-1 is to be installed
       Depends: node-request (>= 2.83~) but 2.26.1-1 is to be installed
       Depends: node-retry (>= 0.10.1~) but 0.6.0-1 is to be installed
       Depends: node-rimraf (>= 2.6.2~) but 2.5.4-2 is to be installed
       Depends: node-semver (>= 5.5~) but 5.3.0-1 is to be installed
       Depends: node-sha (>= 2.0.1~) but 1.2.3-1 is to be installed
       Depends: node-slide (>= 1.1.6~) but 1.1.4-1 is to be installed
       Depends: node-strip-ansi (>= 4.0~) but 3.0.1-1 is to be installed
       Depends: node-tar (>= 4.4~) but 2.2.1-1 is to be installed
       Depends: node-boxen (>= 1.2.1~) but it is not going to be installed
       Depends: node-latest-version (>= 3.0~) but it is not going to be installed
       Depends: node-which (>= 1.3~) but 1.2.11-1 is to be installed
E: Unable to correct problems, you have held broken packages.
```
With that many packages needed for the backport I think it doesn't make sense to try to bring the npm backport into our repository.

nuclearsandwich, 2019-01-18:
Agreed. I think the other option would be to turn on the firehose and enable stretch-backports for the buildfarm, although I don't know how we would document that for downstream rosdep consumers.

clalancette, 2019-01-18:
For what it is worth, it is also not possible to install npm alongside ROS packages on Bionic (at least, as far as I can tell).  On my Bionic container, trying to `apt-get install npm` fails with broken packages.  If I follow the chain back, it is because it wants to install `libssl1.0-dev`.  The problem is that all of the ROS packages (`ros-melodic-*`, `ros-bouncy-*`, `ros-crystal-*`) seem to depend on libssl1.1-dev.  That means that this key isn't going to work on Ubuntu either.

At this point, I'm wondering if it might make more sense to remove the `npm` rosdep key.  I know this might have knock-on effects for any packages depending on it, but it doesn't seem like a viable key to me.

tfoote, 2019-01-19:
Yeah, if it's not compatible with our packages removing the key does make sense. I'd like to have a policy of everything being required to be side by side installable in the database but don't have any way to test or enforce that.

NickZ, 2019-01-19:
@clalancette I am currently trying to fix that issue on Ubuntu; it seems they introduced this dependency issue in the `bionic-updates` version `8.10.0~dfsg-2ubuntu0.4`. If you change the `nodejs` dependency on bionic to be the one in `bionic-security`, which is version `8.10.0~dfsg-2ubuntu0.2`, it depends on `libssl-dev` and this issue is resolved.

sevenbitbyte, 2019-01-22:
Totally understand the challenge of debian/ubuntu + node. I've not had a chance to review dependencies but I wonder if the nodesource -> https://github.com/nodesource/distributions deb packages have the same libssl dependencies. There are many versions to choose from.

Though I think if this does present too much of an OS packaging issue we can instead post this package to npm.

Looking at the success of [rosnodejs](https://www.npmjs.com/package/rosnodejs) with over 600+ weekly downloads its clear the community has already begun adoption of nodejs.

I think going forward the pressures to fully enable nodejs as a development route for ROS developers is here and growing. I fully support catkin/ament as our go to packing tools and really don't want to see a fragmented packaging landscape with packages ending up in language specific repositories. While OK for developers thats not great for user experience.

dirk-thomas, 2020-09-01:
Closing with Stretch being EOL since July 6th, 2020.

closed 2020-09-01 by dirk-thomas