ros/rosdistro #20184: pluginlib doesn't compile on debian:stretch

2ros0, 2019-02-05:
ROS distro: lunar
Debian: stretch
Default boost version through apt: 1.62

Building ROS Lunar from source as directed here: http://wiki.ros.org/Installation/Source. At the step `./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release`, during compilation of `pluginlib`, the executable `plugin_tool` fails at the linking step with the following error:

> librosconsole.so: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform[abi:cxx11](char const*, char const*) const'
> librosconsole.so: undefined reference to `boost::re_detail::raise_runtime_error(std::runtime_error const&)'
> librosconsole.so: undefined reference to `boost::re_detail::put_mem_block(void*)'
> librosconsole.so: undefined reference to `boost::re_detail::perl_matcher<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > >, std::allocator<boost::sub_match<__gnu_cxx::__normal_iterator<char const*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > >, boost::regex_traits<char, boost::cpp_regex_traits<char> > >::construct_init(boost::basic_regex<char, boost::regex_traits<char, boost::cpp_regex_traits<char> > > const&, boost::regex_constants::_match_flags)'
> librosconsole.so: undefined reference to `boost::re_detail::get_mem_block()'
> librosconsole.so: undefined reference to `boost::re_detail::verify_options(unsigned int, boost::regex_constants::_match_flags)'
> librosconsole.so: undefined reference to `boost::re_detail::get_default_error_string(boost::regex_constants::error_type)'
> librosconsole.so: undefined reference to `boost::re_detail::cpp_regex_traits_implementation<char>::transform_primary[abi:cxx11](char const*, char const*) const'
> collect2: error: ld returned 1 exit status
> 

Initially I thought it was a bug in pluginlib/rosconsole. But on looking at other similar cases with other libraries, it seems like it could be a version configuration bug. Is it possible the default version of boost is higher than what is required?

2ros0, 2019-02-05:
Confirmed it works with boost 1.54, so between that and 1.62 something was added that breaks the build. Maybe the rosdep file should be modified to include a specific version number for debian.

nuclearsandwich, 2019-02-05:
@2ros0 what does your workspace contain? I was just able to build pluginlib in a stretch container following the instructions with three modifications:
* Installed rosdep from pip to circumvent an error
* Applied `--rosdistro lunar --os=debian:stretch` to my rosdep invocation since my container wasn't being auto-detected.
* Built with  `./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release --only-pkg-with-deps pluginlib` to build only up to pluginlib.

2ros0, 2019-02-06:
Was a case of multiple installed boost versions. The remote device had an older version of boost preinstalled with some cmake files causing a configuration error. apt's libboost-all-dev (currently 1.62) installed through rosdep works fine. Thanks for checking.

closed 2019-02-06 by 2ros0