ros/rosdistro #20220: Installing ROS Melodic on Ubuntu 18.10

Blind0ne, 2019-02-07:
Is there a way or workaround to install ROS on Ubuntu 18.10? I found [this](https://stackoverflow.com/questions/53266574/installing-ros-melodic-on-ubuntu-18-10) question around that topic which installs it from source and then deals with dependencies and nothing more. I'm new to ROS and don't want to use a virtual machine. Thanks in advance.

PS: Hope this is the right place to post my question.

nuclearsandwich, 2019-02-07:
> PS: Hope this is the right place to post my question.

We ask that questions be directed to http://answers.ros.org 2 as per the FAQ 1 and following our [Support policy](http://wiki.ros.org/Support)

ROS Melodic is only supported on the platforms listed in [REP 3](http://www.ros.org/reps/rep-0003.html#melodic-morenia-may-2018-may-2023). Building from source on unsupported platforms is the recommended path. The StackOverflow question you linked appears to have an answer describing source patches necessary to build on 18.10.

closed 2019-02-08 by tfoote

moriarty, 2019-03-27:
@Blind0ne another solution is just working inside of one of the official docker images (which are based on 18.04) and running that on 18.10

PS: Sorry to bring up an old closed issue- I meant to search closed PRs for something else.