ros/rosdistro #20284: Import libccd-dev v2.1

wxmerkt, 2019-02-12:
The `libccd-dev` package released for all current Debian/Ubuntu distros is `2.0` from ~2014. In the meantime many bug-fixing-but-API-compatible changes have been made - particularly addressing corner cases that could lead to segmentation faults. 

Recently, a new version containing these bugfixes was released (named `2.1` albeit with no changes in the API) which will be part of Debian Buster/Ubuntu Disco:
- https://packages.debian.org/source/buster/libccd
- https://launchpad.net/ubuntu/+source/libccd

I think it'd be useful to provide the new version of `libccd-dev` to ROS users (at least Melodic, potentially Kinetic). A full overview of all changes can be found here:
- https://github.com/danfis/libccd/compare/v2.0...v2.1

A related discussion (thanks to @j-rivero for all the advice) regarding this can be found here: https://github.com/danfis/libccd/issues/42

tfoote, 2019-02-12:
Is there somewhere that these packages are being built as backports? If it really is just a bugfix we can consider it. But someone needs to be packaging it preferably in a public PPA for us to be able to import it.

In general we don't want to get into backporting dependencies that people might rely upon as we will then have to backport it to all supported platforms, and it will add a burden on all users working on community supported platforms.

tfoote, 2019-04-19:
If there's a maintained PPA with these packages to import from we can consider reopening this. But for now I'm going to close it as it's not something that we can effectively resolve here.

closed 2019-04-19 by tfoote