ros/rosdistro #20366: FFMPEG missing from rosdep for Xenial and Bionic

rsterner-hmi, 2019-02-22:
The rosdep entry for ffmpeg is missing itself for Xenial, Bionic and others.  When our package tried to resolve ffmpeg as part of rosdeps on these versions of Ubuntu we were unable to resolve it and had to add the entry in our own rosdep file.  Ffmpeg does exist under other distributions and Ubuntu versions.

tfoote, 2019-02-26:
There's been quite a bit of drama on the ffmpeg vs avcov I guess we followed the Ubuntu/debian maintainers choice to switch. https://github.com/ros/rosdistro/pull/3708

But it sounds like the recommendation is to switch back now: https://superuser.com/questions/507386/why-would-i-choose-libav-over-ffmpeg-or-is-there-even-a-difference but we have been just maintaining. 

What might make sense it to backfill ffmpeg and install it in parallel with the current keys with avcov and then for future rules it will just continue to use the ffmpeg only. This will avoid breaking any existing users, but will get us back on track using the mainline ffmpeg.

nuclearsandwich, 2019-03-06:
Closing now that #20474 is merged.

closed 2019-03-06 by nuclearsandwich