ros/rosdistro #20407: Pre-release tests on Travis passing, but build fails on Jenkins

musamarcusso, 2019-02-27:
Hi,

I have been having the issue for a while that my Travis CI is configured with PRE_RELEASE=true and passing (like this [one](https://travis-ci.org/uuvsimulator/uuv_simulator/builds/498721516)) but failing on Jenkins. The same happens with the [local pre-release test](http://prerelease.ros.org/). 
I am wondering if they don't use the same Docker image or if I am missing some configuration.
Is there a way to set up the build on Travis with the [industrial_ci](https://github.com/ros-industrial/industrial_ci) configuration so that I get the errors and can fix them all before the release? I ran out of ideas to solve this.

tfoote, 2019-02-28:
You can reproduce all the jobs that run on Jenkins https://github.com/ros-infrastructure/ros_buildfarm/blob/master/doc/jobs/release_jobs.rst#run-the-release-job-locally

The most common errors are things that cannot be reproduced efficiently in a prerelease test. For example binary builds are fully isloated, and build on top of the installed versions of all the sub packages. To make that happen and validate you need to run the full release, run bloom, generate all artifacts, build source debs, build binary debs, install, and repeat for each topological sequential build of packages with dependencies.

If you have a specific issue we can try to extend the devel or prerelease tests to cover the issue, but in general there will be things that cannot be efficiently tested without actually running the build. A good place to open the issue(s) is here: https://github.com/ros-infrastructure/ros_buildfarm/issues

closed 2019-02-28 by tfoote

musamarcusso, 2019-02-28:
Thank you, @tfoote. I really appreciate the help.