ros/rosdistro #20591: libgtkmm install error on Ubuntu

Tiryoh, 2019-03-18:
There was an error while installing `libgtkmm` on Ubuntu Xenial + ROS Kinetic.
It is defined to install `libgtkmm-dev` on Ubuntu, but it seems like there are only `libgtkmm-2.4-dev` and `libgtkmm-3.0-dev` on Ubuntu repository (and maybe on Debian, too.)

https://github.com/ros/rosdistro/blob/5b3b503bcdef5827853b20f9efc145012c603496/rosdep/base.yaml#L2059-L2064

We think the rules should be defined as `libgtkmm` and `libgtkmm3`, or `libgtkmm2` and `libgtkmm3`.

* libgtkmm2
  * arch: https://www.archlinux.org/packages/extra/x86_64/gtkmm/
  * debian: https://packages.debian.org/stretch/libgtkmm-2.4-dev
  * fedora: https://apps.fedoraproject.org/packages/gtkmm24-devel
  * gentoo: https://packages.gentoo.org/packages/dev-cpp/gtkmm
  * ubuntu: https://packages.ubuntu.com/xenial/libgtkmm-2.4-dev
* libgtkmm3
  * arch: https://www.archlinux.org/packages/extra/x86_64/gtkmm3/
  * debian: https://packages.debian.org/stretch/libgtkmm-3.0-dev
  * fedora: https://apps.fedoraproject.org/packages/gtkmm30-devel
  * ubuntu: https://packages.ubuntu.com/xenial/libgtkmm-3.0-dev

## Console log

```
$ LANG=C rosdep install -r -y --from-paths .
executing command [sudo -H apt-get install -y libgtkmm-dev]
Reading package lists... Done
Building dependency tree
Reading state information... Done
E: Unable to locate package libgtkmm-dev
ERROR: the following rosdeps failed to install
apt: command [sudo -H apt-get install -y libgtkmm-dev] failed
apt: Failed to detect successful installation of [libgtkmm-dev]
```
```
$ LANG=C sudo apt search libgtkmm-
Sorting... Done
Full Text Search... Done
libgtkmm-2.4-1v5/xenial 1:2.24.4-2 amd64
  C++ wrappers for GTK+ (shared libraries)

libgtkmm-2.4-dbg/xenial 1:2.24.4-2 amd64
  C++ wrappers for GTK+ (debug symbols)

libgtkmm-2.4-dev/xenial 1:2.24.4-2 amd64
  C++ wrappers for GTK+ (development files)

libgtkmm-2.4-doc/xenial,xenial 1:2.24.4-2 all
  C++ wrappers for GTK+ (documentation)

libgtkmm-3.0-1v5/xenial,now 3.18.0-1 amd64 [installed]
  C++ wrappers for GTK+ (shared libraries)

libgtkmm-3.0-dbg/xenial 3.18.0-1 amd64
  C++ wrappers for GTK+ (debug symbols)

libgtkmm-3.0-dev/xenial 3.18.0-1 amd64
  C++ wrappers for GTK+ (development files)

libgtkmm-3.0-doc/xenial,xenial 3.18.0-1 all
  C++ wrappers for GTK+ (documentation)
```

tfoote, 2019-03-19:
Indeed this looks like an invalid rosdep rule. I don't see any packages listing as using it currently in a rosdistro: https://index.ros.org/d/libgtkmm/

It looks to have been one of our early keys, but I can't find a reference to it in the ubuntu old-releases repositories. 

It was refactored in #472 and looks to have initially been added in #358

```
b00429d29 (ipa-fmw                         2013-01-17 10:48:59 +0100 2063) libgtkmm:
dc8320f15 (Paul Hollensen                  2013-07-24 18:27:18 -0300 2064)   arch: [gtkmm]
453b138b5 (Paul Mathieu                    2013-02-13 15:35:14 -0800 2065)   debian: [libgtkmm-dev]
dbfca533c (Ankur Sinha (Ankur Sinha Gmail) 2013-08-05 14:34:50 +1000 2066)   fedora: [gtkmm24]
971b90f1a (Hunter Allen                    2017-06-11 09:34:09 -0700 2067)   gentoo: [dev-cpp/gtkmm]
4f2598fc7 (Paul Mathieu                    2013-02-13 15:08:23 -0800 2068)   ubuntu: [libgtkmm-dev]
```
 In what use case are you running into this issue?

Tiryoh, 2019-03-27:
Sorry for my late reply.

The reason I reported this issue is that we got an error while installing a camera viewer based on GTK+ which contains camera driver.
It seems that the problem is caused by an error installing libgtkmm with rosdep command on the current rosdep rule.

The camera viewer (and the camera driver) is built for a specific camera (it is not public for now), and we are trying to use with ROS.
I think we should use rqt instead of GTK+ in the future and I'm planning it, but it requires GTK+ for now.

tfoote, 2019-03-27:
If you could open a PR with the updated rosdep rule for 2.4 if you're using that or add one for `libgtkmm-3.0-dev` if you're using that one it would be appreciated.

Tiryoh, 2019-03-28:
OK, we're using `libgtkmm-2.4-dev` so I'll open a PR with the updated rosdep rule for 2.4.

Tiryoh, 2019-03-30:
The related PR is merged by 4debd3742eb7d2ae497716c8a218656588c45ced

closed 2019-03-30 by Tiryoh