ros/rosdistro #20600: pip install --upgrade catkin_pkg_module fails

m090009, 2019-03-19:
So I have a Python 2.7 conda environment that I'm trying to run ros (Melodic) from, It worked well but  when I try and run ` pip install --upgrade catkin_pkg_module ` to update and add new modules I get this error:
```
File "/home/m090009/anaconda3/envs/ross/lib/python2.7/site-packages/pip/_internal/wheel.py", line 428, in move_wheel_files
    assert info_dir, "%s .dist-info directory not found" % req
AssertionError: catkin-pkg-modules .dist-info directory not found
```
So I tried to run it outside conda, but I got the same error.

PS: Here are the steps I go through every time Ros code is updated:

- `apt-get update`
- `cd PROJECT_DIR`
- `pip install -r requirements.txt`
- `apt-get install ros-kinetic-dbw-mkz
- `apt-get install ros-kinetic-dbw-mkz-msgs`
- `pip install --upgrade catkin_pkg_modules`

tfoote, 2019-03-19:
Since anaconda is not a supported platform/environment I'm going to close this.

That said we're happy to help you debug, but since this is relatively specific to anaconda and anaconda does things differently than standard python environments it's not something that we're going to debug. If you want help debugging please provide enough information to reproduce your full environment on a question on answers.ros.org. The shorter and more compact you can make it to reproduce the issue the more likely you are to get help.

And if you find the problem, a PR against [catkin_pkg](https://github.com/ros-infrastructure/catkin_pkg) which is where the modules is hosted would be appreciated.

closed 2019-03-19 by tfoote

m090009, 2019-03-19:
Hi @tfoote , my bad the problem was in part due too anaconda. However, the real issue is after running `catkin_make` I got 
```
Traceback (most recent call last):
  File "/opt/ros/melodic/share/gencpp/cmake/../../../lib/gencpp/gen_cpp.py", line 49, in <module>
    srv_template_map)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 213, in generate_from_command_line_options
    generate_from_file(argv[1], options.package, options.outdir, options.emdir, options.includepath, msg_template_dict, srv_template_dict)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 154, in generate_from_file
    _generate_msg_from_file(input_file, output_dir, template_dir, search_path, package_name, msg_template_dict)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 99, in _generate_msg_from_file
    search_path)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/genmsg/template_tools.py", line 77, in _generate_from_spec
    interpreter = em.Interpreter(output=ofile, globals=g, options={em.RAW_OPT:True,em.BUFFERED_OPT:True})
AttributeError: 'module' object has no attribute 'Interpreter'

```
error, so I thought it was the result of not running  `pip install --upgrade catkin_pkg_module`, but after switching from `conda` to native Python and successfully running `pip install --upgrade catkin_pkg_module` the problem persisted. 

So I did a little digging and found out that its a conflict between the `em` and the `empy` packages and I simply need to remove the `em` package which I did via `pip uninstall em` and made sure I had `empy`. But I still got the error, and after a little digging and going through the code I found out that python still loads the `em` package from `/usr/lib/python2.7/dist-packages/em.pyc`, so how can I completely remove any trace of em to let python use `empy instead`?

tfoote, 2019-03-20:
The simple solution is to rm it. The better solution is to find out what
installed it and make sure that you have uninstalled it correctly. Such as
seeing if it came from a Debian package that you don't remember installing
etc.