ros/rosdistro #21106: python package pep8 has renamed to pycodestyle

russkel, 2019-05-07:
> pycodestyle is a tool to check your Python code against some of the style conventions in PEP 8.
> 
> Note
> 
> This package used to be called pep8 but was renamed to pycodestyle to reduce confusion. Further discussion can be found in the issue where Guido requested this change, or in the lightning talk at PyCon 2016 by @IanLee1521: slides video.

Should the references to this dependency be changed to `pycodestyle` in the code base?

tfoote, 2019-05-07:
It would be good to migrate to this, but we'll need to have a migration plan and deprecation cycle. Probably adding a new key with the new name and keeping the old key with the old name on existing platforms (but disabled on new platforms) so that people will switch over.

dirk-thomas, 2020-09-01:
Closing due to age and inactivity.

closed 2020-09-01 by dirk-thomas